RT-Thread/rt-thread #7959: mbedtls-v2.7.10.2 fatal error: sys/ioctl.h: No such file or directory

zhjc2014, 2023-08-17:
芯片：stm32f103cb
工具链：rt-studio 2.2.7
平台：win10
rtthread settings里面使能webclient软件包 配置如截图所示 
[image]
编译错误输出 
[image]
../packages/mbedtls-v2.7.10.2/ports/src/tls_net.c:76:23: fatal error: sys/ioctl.h: No such file or directory

mysterywolf, 2023-08-17:
请问rt-thread是哪个版本的？

zhjc2014, 2023-08-26:
> 请问rt-thread是哪个版本的？

4.0.3
[image]

zhenbin-lin, 2024-06-24:
我也遇到这个问题了，请问怎么解决的呢？

1078249029, 2025-03-06:
esp32 gcc Ubuntu22.04版本遇到同样的问题，可以通过绝对地址的方式导入头文件，但不能通过编译器自行查找的方式找到头文件，否则自动导入本机(usr/x86_64)的ioctl.h文件