RT-Thread/rt-thread #7964: qemu debug rt-thread error

TaiJuWu, 2023-08-20:
rt-thread commit: 166c83f37
bsp: qemu-vexpress-a9
toolchain: arm-none-eabi-gcc (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
QEMU version: QEMU emulator version 5.2.0 (Debian 1:5.2+dfsg-11+deb11u2)

使用qemu-nographic.sh沒有問題，但是用了QEMU-dbg.sh就會報下面錯誤
```
WARNING: Image format was not specified for 'sd.bin' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM default
alsa: Could not initialize DAC
alsa: Failed to open `default':
alsa: Reason: No such file or directory
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM default
alsa: Could not initialize DAC
alsa: Failed to open `default':
alsa: Reason: No such file or directory
audio: Failed to create voice `lm4549.out'
QEMU: Terminated
```

BernardXiong, 2023-09-03:
好像不会，

或者你可以对qemu-nographic.sh做简单的调整，在qemu-system-arm的命令行后面加入`-S -s`

TaiJuWu, 2023-09-07:
> 好像不会，
> 
> 或者你可以对qemu-nographic.sh做简单的调整，在qemu-system-arm的命令行后面加入`-S -s`来让qemu进入调试等待模式

這個方法試過了哦

BernardXiong, 2023-09-09:
> 這個方法試過了哦

那这个就有些不清楚了呢，可能需要自己仔细看看环境相关的因素。