RT-Thread/rt-thread #7982: AARCH64 Qemu 关闭 SMP 时编译不通过

zhugengyu, 2023-08-25:
- AARCH64 的几个 BSP 都必须打开 SMP 才能编译通过，否则会提示错误

```
/home/zhugy/Project/rt-thread/work/libcpu/aarch64/common/context_gcc.S: Assembler messages:
/home/zhugy/Project/rt-thread/work/libcpu/aarch64/common/context_gcc.S:538: Error: unknown mnemonic `save_user_ctx' -- `save_user_ctx'
/home/zhugy/Project/rt-thread/work/libcpu/aarch64/common/context_gcc.S:554: Error: unknown mnemonic `save_user_ctx' -- `save_user_ctx'
scons: *** [build/kernel/libcpu/aarch64/common/context_gcc.o] Error 1
scons: building terminated because of errors.
```

BernardXiong, 2023-09-03:
谢谢，欢迎提交PR来修正这个问题。或者，稍微等待一段时间，我们会修正这个问题。

mysterywolf, 2023-09-16:
https://github.com/RT-Thread/rt-thread/pull/8045

closed 2023-09-16 by mysterywolf