RT-Thread/rt-thread #7985: I2C相关PR整合

mysterywolf, 2023-08-25:


mysterywolf, 2023-10-04:
1. 软件i2c目前套层太多
2. rt_i2c_bit_ops 结构体中超时(.timeout)和延时(.delay_us)，单位不统一 应该统一为us
3. 取消us 方法接口

sp-cai, 2023-10-18:
> 1. 软件i2c目前套层太多
> 2. rt_i2c_bit_ops 结构体中超时(.timeout)和延时(.delay_us)，单位不统一 应该统一为us
> 3. 取消us 方法接口

1.开始时也是觉得套层有点多，去除了 i2c-bit-ops.c ，后来 @supperthomas 建议保留原来的分层结构，于是又改了回去
2.超时(.timeout)是通过系统 tick 计时的，可以在 i2c_cfg 时转换一下
3.这点不明白，如果取消了us方法，用什么替代？

supperthomas, 2023-10-18:
> > 1. 软件i2c目前套层太多
> > 2. rt_i2c_bit_ops 结构体中超时(.timeout)和延时(.delay_us)，单位不统一 应该统一为us
> > 3. 取消us 方法接口
> 
> 1.开始时也是觉得套层有点多，去除了 i2c-bit-ops.c ，后来 @supperthomas 建议保留原来的分层结构，于是又改了回去 2.超时(.timeout)是通过系统 tick 计时的，可以在 i2c_cfg 时转换一下 3.这点不明白，如果取消了us方法，用什么替代？

感谢贡献，请问可以在论坛里介绍下如何使用和一些其他bsp使用的时候需要的注意事项吗？https://club.rt-thread.org/index.html