RT-Thread/rt-thread #7991: scons 4.5.2 build error

Guozhanxin, 2023-08-29:
新版本scons编译报错
[image]
注释掉下面这行代码可以正常使用
[image]

zhkag, 2023-09-01:
这就是原来 scons 固定到 4.4.0 的原因 后来熊大修了，高版本的 scons  2023-5-22 之前的版本都不好使的

BernardXiong, 2023-09-03:
> 这就是原来 scons 固定到 4.4.0 的原因 后来熊大修了，高版本的 scons 2023-5-22 之前的版本都不好使的

这个是说修出问题了吗？

zhkag, 2023-09-05:
感觉这个问题提到长期支持版中，重新释放一下版本吧，随着时间的推移肯定越来越多的人会更新到 4.4.0 之后的 scons 版本，到时候都会有这个问题的

Guozhanxin, 2023-09-06:
> 感觉这个问题提到长期支持版中，重新释放一下版本吧，随着时间的推移肯定越来越多的人会更新到 4.4.0 之后的 scons 版本，到时候都会有这个问题的

哪一笔代码修复的啊

zhkag, 2023-09-07:
> > 感觉这个问题提到长期支持版中，重新释放一下版本吧，随着时间的推移肯定越来越多的人会更新到 4.4.0 之后的 scons 版本，到时候都会有这个问题的
> 
> 哪一笔代码修复的啊

#7541

closed 2024-04-20 by BernardXiong