RT-Thread/rt-thread #7996: 4.1.0和5.0.0版本，使用Keil5.38版本编译bsp\gd32450z-eval\project.uvprojx并下载到下载gd32450z-eval开发板，运行异常

newsky79, 2023-08-31:
其中4.1.0上电即异常，打印输出如下：

[09:48:22.840]收←◆
 \ | /
- RT -     Thread Operating System
 / | \     4.1.0 build Aug 31 2023 09:48:05
 2006 - 2022 Copyright by RT-Thread team
(rt_object_get_type(&m->parent) == RT_Object_Class_Memory) assertion failed at function:rt_smem_alloc, line number:289 

5.0.0上电即异常，打印输出如下：

[10:03:28.239]收←◆
 \ | /
- RT -     Thread Operating System
 / | \     5.0.0 build Aug 31 2023 09:44:18
 2006 - 2022 Copyright by RT-Thread team
(rt_object_get_type(&m->parent) == RT_Object_Class_Memory) assertion failed at function:rt_smem_alloc, line number:287

mysterywolf, 2023-08-31:
bsp\gd32450z-eval\ 这个bsp已经废弃掉了 请使用bsp/gd32/文件夹下的

newsky79, 2023-08-31:
bsp\gd32\arm\gd32450z-eval下的project还没有把网口及网络协议栈、console等匹配开发板接口的功能合进去，哪里有能全面合入这些功能的rtthread sdk吗？