RT-Thread/rt-thread #8003: rt-thread项目的cpu架构移植工作难度问题

wangyuliu, 2023-09-05:
各位好！我在实现一个关于CPU架构的移植工作复杂度评估的工具，主要面向riscv架构移植，我收集rt-thread项目作为我的数据集之一，通过扫描工具得出其架构移植复杂度较大，工程中与CPU架构相关的代码量多，需要较专业团队进行，这样的判断是否准确？期待您的帮助与回答

aozima, 2023-09-05:
关联 #8002  ，问题发在同1个issue中就行了。
不用发PR吧？ #8004

要不看下 [/libcpu/risc-v](https://github.com/RT-Thread/rt-thread/tree/v5.0.x/libcpu/risc-v)基本已包含CPU相关的代码的代码了吧？
甚至你要移植的芯片都已经在bsp目录下了。

Guozhanxin, 2023-09-06:
不太准确，其实RT-Thread有个通用的riscv适配层，移植起来还是比较简单的。https://github.com/RT-Thread/rt-thread/tree/master/libcpu/risc-v

wangyuliu, 2023-09-06:
抱歉，我可能没有描述清楚问题，我希望您可以为我提供该工程全部移植工作量与难度的评估，包括已经被团队适配的部分以及可能待适配的全部工作量，也是为了获取指标帮助工具预测其他未移植的软件。例如，团队尚未开发riscv架构功能到完成架构适配是否可能需要较大的工作量呢？我的扫描工具获取的信息是该工程包含许多关于架构的代码，包括大量汇编、内建函数等，所以得出移植难度较高这样的结果。感谢您！

supperthomas, 2023-09-06:
请提供“关于CPU架构的移植工作复杂度评估的工具” 相关内容。

wangyuliu, 2023-09-06:
谢谢，那评估其为中等难度应该较为合适吗

BernardXiong, 2023-09-09:
你至少得要把你的背景情况说明清楚吧。这个看下来有些类似于要提供一份评估报告？类似的估计也只能点到为止的，和真实的要去做一份移植相差甚远。这类也不能说，libcpu下的代码多，所以对应的移植就难。因为软件上，一些代码，一些函数本身提供出来就是考虑复用，从而把移植变得简单。

另外，做为移植来说，专业的人移植起来显然简单。如果只是一个小白，从0开始学习cpu架构，这个移植过程肯定难。

所以这类评估应该说，太主观了。

closed 2023-09-15 by mysterywolf

wangyuliu, 2023-12-05:
> 请提供“关于CPU架构的移植工作复杂度评估的工具” 相关内容。
> 
> 你可以先看下文档： https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/application-note/porting/risc-v/port-risc-v?id=risc-v-%e7%a7%bb%e6%a4%8d 实际上，不同芯片移植难易度不同。 而且关于架构的代码不是很多。基本和freertos差不多。

非常感谢您的回复，我们的工具通过对libcpu/riscv目录下内容的重新验证发现仍然存在千行以上的汇编代码，因此通过自动化评定移植复杂程度为中等，而不是非常简单，这样是正确的吗？期待您的下一步回复，谢谢。我们的工具虽然存在过于主观的问题，但其目的就是在开发者可能不十分熟悉的情况下，对工程的迁移难度有一个初步的认识，并且能够在此基础上合理地规划分配资源等。

mysterywolf, 2023-12-05:
你好 我们甚至都不知道你的软件判定逻辑是什么，根据什么判定是难还是简单还是中等？我注意到你给很多社区都发了类似的评估，他们也有同样的疑问。

mysterywolf, 2023-12-05:
在没有详细解释你的判定方式和细则之前，我们不认可你给出的结论。