RT-Thread/rt-thread #8009: 最新的env script  `pkgs --update` 有问题

supperthomas, 2023-09-07:
```
 ~/work/rt-thread/rt-thread/bsp/nrf5x/nrf51822 ~/work/rt-thread/rt-thread
Traceback (most recent call last):
   File "/home/runner/.env/tools/scripts/env.py", line 34, in <module>
   from cmds import *
   File "/home/runner/.env/tools/scripts/cmds/cmd_menuconfig.py", line 31, in <module>
    import psutil
 ModuleNotFoundError: No module named 'psutil'
```

https://pipelines.actions.githubusercontent.com/serviceHosts/1d3c4c1d-4e0d-4c69-b441-523a6507202c/_apis/pipelines/1/runs/33522/signedlogcontent/14?urlExpires=2023-09-07T01%3A44%3A04.9748193Z&urlSigningMethod=HMACV1&urlSignature=Wnh93iVo6ZU746%2B3z1nJQEw4VJEhSCQV6TxD7k09eXg%3D

mysterywolf, 2023-09-07:
昨天已经修复了 

https://github.com/RT-Thread/env-windows/commit/9ba1a1290a47e2a3c789a3ef0e5df49d78f5c07e

https://github.com/RT-Thread/env/commit/77ad79c582e0c0f3fa60eb5b4d4a7623ab642d84

supperthomas, 2023-09-07:
> 昨天已经修复了
> 
> [RT-Thread/env-windows@9ba1a12](https://github.com/RT-Thread/env-windows/commit/9ba1a1290a47e2a3c789a3ef0e5df49d78f5c07e)
> 
> [RT-Thread/env@77ad79c](https://github.com/RT-Thread/env/commit/77ad79c582e0c0f3fa60eb5b4d4a7623ab642d84)

action 顺便修复一下?

mysterywolf, 2023-09-07:
action 可否考虑用标准的env在ubuntu下的安装方式：
wget https://raw.githubusercontent.com/RT-Thread/env/master/install_ubuntu.sh
chmod 777 install_ubuntu.sh
./install_ubuntu.sh

这个脚本是env自带的 可以自动安装所有依赖并自动搭建环境

supperthomas, 2023-09-07:
> action 可否考虑用标准的env在ubuntu下的安装方式： wget https://raw.githubusercontent.com/RT-Thread/env/master/install_ubuntu.sh chmod 777 install_ubuntu.sh ./install_ubuntu.sh
> 
> 这个脚本是env自带的 可以自动安装所有依赖并自动搭建环境
> 
> 这样action就不用有一套自己的安装环境的流程了

可以呀，可以试试。欢迎PR。

mysterywolf, 2023-09-07:
哪个文件需要修改 涛哥可以提供一下嘛？ 我看看怎么改

supperthomas, 2023-09-07:
https://github.com/RT-Thread/rt-thread/blob/master/.github/workflows/action.yml

supperthomas, 2023-09-07:
https://github.com/RT-Thread/rt-thread/blob/5f948690c9b2190c8a237bdfc0e67f4d68f4efd6/.github/workflows/action.yml#L325

mysterywolf, 2023-09-07:
https://github.com/RT-Thread/rt-thread/pull/8010 涛哥你看一下

closed 2023-09-07 by mysterywolf

supperthomas, 2023-09-07:
```
        run: |
          wget https://raw.githubusercontent.com/RT-Thread/env/master/install_ubuntu.sh
          chmod 777 install_ubuntu.sh
          ./install_ubuntu.sh
          git config --global http.postBuffer 524288000
          python -c "import tools.menuconfig; tools.menuconfig.touch_env()"
          echo "RTT_ROOT=${{ github.workspace }}" >> $GITHUB_ENV
```

mysterywolf, 2023-09-07:
涛哥你来提个pr吧 action这块我不熟悉