RT-Thread/rt-thread #8056: [proposal] `rt_timer_check` 长时间关闭中断，影响系统实时性, 或许可以优化

a1012112796, 2023-09-19:
目前 `rt_timer_check` 的为了确保 timer list 的一致性，全程关闭中断 (时间长达数十us, 且理论上会随定时器数量的增加而变大)，影响了系统的实时性，或许可以通过算法优化，尽可能减少中断关闭的时间。

此问题是社区微信群里的大佬提出的，个人认为确实值得优化，但暂时没想到太好的方案， mark 一下。

https://github.com/RT-Thread/rt-thread/blob/35231eb116c44165d881dd10ff15d27c4ca540d1/src/timer.c#L665

一些 tips: 
- 群里大佬提到了在执行 hook 的时候暂时恢复中断来减少中断时间，但我感觉不太可。。。示例如下: 
```DIFF
@@ -685,9 +685,16 @@ void rt_timer_check(void)
             }
             /* add timer to temporary list  */
             rt_list_insert_after(&list, &(t->row[RT_TIMER_SKIP_LIST_LEVEL - 1]));
+
+            /* enable interrupt */
+            rt_hw_interrupt_enable(level);
+
             /* call timeout function */
             t->timeout_func(t->parameter);
 
+            /* disable interrupt */
+            level = rt_hw_interrupt_disable();
+
             /* re-get tick */
             current_tick = rt_tick_get();
```
- 个人对别的实时操作系统了解不多，好奇 free-rtos 等其它实时操作系统是如何做的，是否有更好的方案.

mysterywolf, 2023-09-19:
related: https://github.com/RT-Thread/rt-thread/issues/7860

Rei-4396, 2023-10-10:
freertos是基于queue实现的，task本身并不依赖于timer，其timer是一个可选项。

freertos对于定时器的处理是，所有的操作都通过push to queue去完成，之后在tmr svc线程中pop from queue依次执行所有的定时器操作，显然不存在竞争关系，因此可以收敛为单线程场景，不需要考虑同步问题，也就不需要关中断实现一个临界区了。

但rt-thread中thread依赖于timer的实现，因此不太好借鉴其实现，但或许可以考虑学习其思想，将timer收敛为单线程场景，这样就能去掉锁了？

a1012112796, 2023-10-11:
[image]

[引用链接](https://blog.csdn.net/XieWinter/article/details/103088356#:~:text=%E4%BB%BB%E5%8A%A1%E5%BB%B6%E6%97%B6%E5%88%97%E8%A1%A8%E7%9A%84%E5%B7%A5%E4%BD%9C%E5%8E%9F%E7%90%86,%E5%9C%A8FreeRTOS%E4%B8%AD%EF%BC%8C%E6%9C%89%E4%B8%80%E4%B8%AA%E4%BB%BB%E5%8A%A1%E5%BB%B6%E6%97%B6%E5%88%97%E8%A1%A8%EF%BC%88%E5%AE%9E%E9%99%85%E4%B8%8A%E6%9C%89%E4%B8%A4%E4%B8%AA%EF%BC%8C%E4%B8%BA%E4%BA%86%E6%96%B9%E4%BE%BF%E8%AE%B2%E8%A7%A3%E5%8E%9F%E7%90%86%EF%BC%8C%E6%88%91%E4%BB%AC%E5%81%87%E8%A3%85%E5%90%88%E5%B9%B6%E4%B8%BA%E4%B8%80%E4%B8%AA%EF%BC%8C%E5%85%B6%E5%AE%9E%E4%B8%A4%E4%B8%AA%E7%9A%84%E4%BD%9C%E7%94%A8%E6%98%AF%E4%B8%80%E6%A0%B7%E7%9A%84%EF%BC%89%EF%BC%8C%E5%BD%93%E4%BB%BB%E5%8A%A1%E9%9C%80%E8%A6%81%E5%BB%B6%E6%97%B6%E7%9A%84%E6%97%B6%E5%80%99%EF%BC%8C%E5%88%99%E5%85%88%E5%B0%86%E4%BB%BB%E5%8A%A1%E6%8C%82%E8%B5%B7%EF%BC%8C%E5%8D%B3%E5%85%88%E5%B0%86%E4%BB%BB%E5%8A%A1%E4%BB%8E%E5%B0%B1%E7%BB%AA%E5%88%97%E8%A1%A8%E5%88%A0%E9%99%A4%EF%BC%8C%E7%84%B6%E5%90%8E%E6%8F%92%E5%85%A5%E5%88%B0%E4%BB%BB%E5%8A%A1%E5%BB%B6%E6%97%B6%E5%88%97%E8%A1%A8%EF%BC%8C%E5%90%8C%E6%97%B6%E6%9B%B4%E6%96%B0%E4%B8%8B%E4%B8%80%E4%B8%AA%E4%BB%BB%E5%8A%A1%E7%9A%84%E8%A7%A3%E9%94%81%E6%97%B6%E5%88%BB%E5%8F%98%E9%87%8F%EF%BC%9AxNextTaskUnblockTime%E7%9A%84%E5%80%BC%E3%80%82)

a1012112796, 2023-11-14:
https://github.com/RT-Thread/rt-thread/pull/8042#discussion_r1391914034

xqyjlj, 2023-12-15:
> [#8042 (comment)](https://github.com/RT-Thread/rt-thread/pull/8042#discussion_r1391914034)

已验证此改动会造成死锁bug，此pr会进行修复 https://github.com/RT-Thread/rt-thread/pull/8369

xqyjlj, 2023-12-15:
rt_timer_check 这个函数也确实要优化了，该函数会严重影响实时性，造成严重抖动。多核上效果更恐怖