RT-Thread/rt-thread #8098: dev_pwm使用后体验反馈建议

meng-plus, 2023-09-26:
使用环境STM32F103, iar开发环境，PWM用作电机控制

1. 按照说明文档配置后 PWM无输出问题，经排查为未配置PWM输出 [[pwm_config.h](https://github.com/meng-plus/rt-thread/blob/efc1a57bc9b0ac5c9f38c6c711418bd473d89b0c/bsp/stm32/libraries/HAL_Drivers/config/f1/pwm_config.h#L26)](https://github.com/meng-plus/rt-thread/blob/efc1a57bc9b0ac5c9f38c6c711418bd473d89b0c/bsp/stm32/libraries/HAL_Drivers/config/f1/pwm_config.h#L26)

   ```c
   #define PWM1_CONFIG                             \
       {                                           \
          .tim_handle.Instance     = TIM1,         \
          .name                    = "pwm1",       \
          .channel                 = RT_NULL       \
       }
   ```

   **.channel** 默认配置全为RT_NULL,而经过排查源码发现此处的配置决定了通道的开关，但是此处的源码为所有STM32单片机公用，也就无法自定义修改，但是只有[[/* config pwm channel */](https://github.com/meng-plus/rt-thread/blob/efc1a57bc9b0ac5c9f38c6c711418bd473d89b0c/bsp/stm32/libraries/HAL_Drivers/drv_pwm.c#L424-L433)](https://github.com/meng-plus/rt-thread/blob/efc1a57bc9b0ac5c9f38c6c711418bd473d89b0c/bsp/stm32/libraries/HAL_Drivers/drv_pwm.c#L424-L433)处有对PWM通道进行配置。
     因此我认为此处设计不合理，建议在调用**rt_pwm_enable**  调用**HAL_TIM_PWM_ConfigChannel**完成PWM口的功能配置。

2. rt_pwm_set((struct rt_device_pwm *)pwm_dev, PWM_DEV_CHANNEL, period, pulse); 传参过多，在使用加减速调节算法中， 只需要占空比参数即可，但是我未在驱动中发现类似的配置，建议考虑下将这个接口功能拆分以提高效率。

mysterywolf, 2023-10-04:
可否把代码粘过来看一下

meng-plus, 2023-10-04:
> 可否把代码粘过来看一下

还没把功能整理好看，后续整理清楚了 给提交个pr

closed 2024-12-05 by Rbb666