RT-Thread/rt-thread #8116: rt_thread中如何使用TrustZone

li301597, 2023-10-08:
rt_thread中如何使用arm CM33的内核的TrustZone？

enkiller, 2023-11-21:
rt_thread 系统运行在非安全态，打开 ARM_CM33_ENABLE_TRUSTZONE 后，需要**安全态的库**提供相关的API，用于支持安全态线程上下文保存功能。

li301597, 2023-11-21:
请问下”安全态的库提供相关的API”可以从哪里获取，不会是要自己实现吧？

enkiller, 2023-11-22:
类似 `TZ_StoreContext_S` 的函数可以看看 ARM 给的示例：

[Examples/Fault/B-U585I-IOT02A/Secure/tz_context.c](https://github.com/ARM-software/CMSIS-View/blob/main/Examples/Fault/B-U585I-IOT02A/Secure/tz_context.c)

TrustZone 一般是跑加解密，鉴权等操作的。还需要搭配类似 `trusted-firmware-m` 的库一起使用

closed 2024-12-05 by Rbb666

suntong30, 2025-03-22:
> 类似 `TZ_StoreContext_S` 的函数可以看看 ARM 给的示例：
> 
> [Examples/Fault/B-U585I-IOT02A/Secure/tz_context.c](https://github.com/ARM-software/CMSIS-View/blob/main/Examples/Fault/B-U585I-IOT02A/Secure/tz_context.c)
> 
> TrustZone 一般是跑加解密，鉴权等操作的。还需要搭配类似 `trusted-firmware-m` 的库一起使用

请问有具体的rt-thread和tf-m交互的案例吗？