RT-Thread/rt-thread #8135: rt-tread4.1.1开启C++支持之后，全局对象的析构函数不会被调用

atwwww, 2023-10-14:
### 环境：
windows 10
RT-Thread Studio 2.7.7

### 编译器：
arm-none-eabi-gcc 5.4.1
### 工程信息
[image]

### 问题复现：
1. 打开C++支持
2. 在main函数外声明一个全局对象
3. 运行完毕之后发现，全局对象的构造函数被调用了，但**全局对象的析构函数没有被调用**。
代码
[image]
结果
[image]

### 目前已知：
1. RT-Thread全局对象的构造函数会在链接脚本机制下被收集放到.ctors段中，在函数cplusplus_system_init中被调用，以此来初始化。
2. RT-Thread链接脚本中似乎也有析构函数对应的段，但是似乎没有被使用。[image]
3. 在Ubuntu20.04中，发现main函数结束后会跳入exit函数中，在此函数中，全局对象的析构函数被执行；但是RT-Thread中，main函数结束之后，直接跳到_thread_exit函数，将main线程detach。调度到idle线程时，清除main线程有关资源。
5. 全程没有调用到全局对象的析构函数。
6. 局部对象的构造函数和析构函数可以正常调用。

### 想要了解
1. 什么原因导致的全局对象的析构函数没有被调用?
2. 怎样才能够补全这个全局对象的自动析构机制?

JonasWen, 2023-10-16:
是不是作用域的问题。你把对象声明放到main里去就会调用析构函数了

atwwww, 2023-10-16:
> 是不是作用域的问题。你把对象声明放到main里去就会调用析构函数了

放到main中确实可以，相当于是main的局部对象。编译器产生的汇编文件中是可以看到在函数结束前调用对象的析构函数的。
但是放到main外，作为一个真正的全局变量的时候似乎只能初始化，但是最后不会被调用析构函数。
我在问题中也说明了这种情况：
[image]

JonasWen, 2023-10-16:
你调用exit(0);试试看

atwwww, 2023-10-17:
> 你调用exit(0);试试看

之前也试过手动调用exit(0)，然后又尝试了一次。
RT-THREAD实现了newlib中的exit的桩函数_exit，实现方式是将当前线程直接delete掉🤔
[image]

[image]

[image]

JonasWen, 2023-10-17:
你有keil的环境吗，我用MDK5.38/ArmCompiler6.19编译后调用exit(0);会执行全局变量的析构函数。

atwwww, 2023-10-17:
> 你有keil的环境吗，我用MDK5.38/ArmCompiler6.19编译后调用exit(0);会执行全局变量的析构函数。

我用的gnu的编译器，然后就用的RT-Thread Studio

atwwww, 2023-10-17:
> 你有keil的环境吗，我用MDK5.38/ArmCompiler6.19编译后调用exit(0);会执行全局变量的析构函数。

我发现链接器配置参数，不要使用简化的newlib库就可以了。
[image]

[image]

closed 2023-10-17 by atwwww

atwwww, 2023-10-24:
前段时间研究发现，nano-newlib中使用的应该是轻量级exit，轻量级退出程序在程序退出时不会清理全局变量。
在newlib的README中：
[image]

在newlib源码exit函数中：
[image]

__atexit.c中关于lite_exit说明
[image]