RT-Thread/rt-thread #8153: AT32UC3B: rt_strerror(): for loop initial declaration: Compilation error

ramangopalan, 2023-10-18:
Dear RT-Thread community,

During the invocation of scons on the BSP for at32uc3b, the
compilation fails at kservice.c. Here is the gist of the problem.

CC build\kernel\src\kservice.o
C:\Skullets\rt-thread\rt-thread\src\kservice.c: In function 'rt_strerror':
C:\Skullets\rt-thread\rt-thread\src\kservice.c:136: error: 'for' loop initial declarations are only allowed in C99 mode
C:\Skullets\rt-thread\rt-thread\src\kservice.c:136: note: use option -std=c99 or -std=gnu99 to compile your code
scons: *** [build\kernel\src\kservice.o] Error 1
scons: building terminated because of errors.

For now (and in order to get things moving quickly), I locally moved
the iterator's declaration outside the for statement. A quick glance
at other BSP directories tells me that this style of declaration
(which requires -std=gnu99) isn't used extensively.

What would be a good fix? Should I fix this by forcing gnu99 within
the BSP python configuration file OR move the iterator symbol to the
top of the rt_strerror()?

mysterywolf, 2023-10-22:
Hi, thanks for you report, 
Could you submit a PR please:

```diff
diff --git a/src/kservice.c b/src/kservice.c
index 5b005a9551c511c5363d85b11370f6c352cab922..c1b7146dd59d896ef1b826e5308a9654ff9ceedd 100644
--- a/src/kservice.c
+++ b/src/kservice.c
@@ -130,10 +130,12 @@ static struct _errno_str_t  rt_errno_strs[] =
  */
 const char *rt_strerror(rt_err_t error)
 {
+    int i = 0;
+
     if (error < 0)
         error = -error;
 
-    for (int i = 0; i < sizeof(rt_errno_strs) / sizeof(rt_errno_strs[0]); i++)
+    for (i = 0; i < sizeof(rt_errno_strs) / sizeof(rt_errno_strs[0]); i++)
     {
         if (rt_errno_strs[i].error == error)
             return rt_errno_strs[i].str;
```

ramangopalan, 2023-10-23:
Perfect! That solves the problem. Thank you @mysterywolf!

closed 2023-10-23 by mysterywolf