RT-Thread/rt-thread #8156: [bsp] [ESP32_C3] [main/CMakeLists.txt] 源列表里有些文件已变更路径或拆分了

sp-cai, 2023-10-19:
在 bsp/ESP32_C3/main/CMakeLists.txt 发现有以下文件找不到：
        ../../../src/device.c
https://github.com/RT-Thread/rt-thread/blob/d01dd05a0cd2107d85e26a445a43957880fd1978/bsp/ESP32_C3/main/CMakeLists.txt#L23C23-L23C23

        ../../../src/scheduler.c
https://github.com/RT-Thread/rt-thread/blob/d01dd05a0cd2107d85e26a445a43957880fd1978/bsp/ESP32_C3/main/CMakeLists.txt#L31C26-L31C26

原因是原来的源文件已变更路径或已拆分了

tangzz98, 2023-10-20:
现在已经不用CMake构建了 #7821

sp-cai, 2023-10-20:
> 现在已经不用CMake构建了 #7821

不再用的文件也可以清理一下的

tangzz98, 2023-10-20:
@WCX1024979076

WCX1024979076, 2023-10-20:
这个esp32的编译不在再采用esp-idf的idf.py build，目前采用scons编译，所以cmake.txt没再维护，具体参考最新的readme

WCX1024979076, 2023-10-20:
这个我提个pr把cmake.txt删了吧

tangzz98, 2023-10-20:
> 这个我提个pr把cmake.txt删了吧

Makefile, rtt.patch都可以不要了。sdkconfig不知道还有没有用？

WCX1024979076, 2023-10-20:
> > 这个我提个pr把cmake.txt删了吧
> 
> Makefile, rtt.patch都可以不要了。sdkconfig不知道还有没有用？
sdkconfig也没啥用了，删了吧

closed 2023-10-22 by mysterywolf