RT-Thread/rt-thread #8283: Weak random source in RT-Thread rt_random driver

0xdea, 2023-11-24:
Hi,

I would like to report another potential vulnerability in the current version of RT-Thread. Please let me know if you plan to ask for a CVE ID in case the vulnerability is confirmed. I'm available if you need further clarifications.

## Weak random source in RT-Thread rt_random driver

### Summary
I noticed a weak random source at the following location in the RT-Thread rt_random driver source code:
https://github.com/RT-Thread/rt-thread/blob/master/components/drivers/misc/rt_random.c#L16-L28

### Details
Weak random generator in the the `calc_random()` function:
```c
static rt_uint16_t calc_random(void)
{
    seed = 214013L * seed + 2531011L;
    return (seed >> 16) & 0x7FFF;   /* return bits 16~30 */
}

static rt_ssize_t random_read(rt_device_t dev, rt_off_t pos, void *buffer, rt_size_t size)
{
    rt_uint16_t rand = calc_random(); /* VULN: weak random source */
    ssize_t ret = sizeof(rand);
    rt_memcpy(buffer, &rand, ret);
    return ret;
}
```

### Impact
If the reported weak random generator is used in a security context, such as cryptographic operations, it can expose to a number of attacks aimed to break encryption.

0xdea, 2023-12-24:
Hi, it's been one month since I reported this vulnerability, and I wanted to ask if you have any update. As standard practice, I plan to request a CVE ID for every confirmed vulnerability. I also intend to publish an advisory by February at the latest, unless there's a specific reason to postpone. Thanks!

0xdea, 2024-02-08:
Hi there, CVE-2024-25389 was assigned to this vulnerability. I'm planning to publish my security advisory and writeup on March 5th. Thanks.

cms19859230182-lang, 2026-11-10:
/claim