RT-Thread/rt-thread #8285: Potential heap buffer overflow in RT-Thread wlan driver

0xdea, 2023-11-24:
Hi,

I would like to report another potential vulnerability in the current version of RT-Thread. Please let me know if you plan to ask for a CVE ID in case the vulnerability is confirmed. I'm available if you need further clarifications.

## Potential heap buffer overflow in RT-Thread wlan driver

### Summary
I spotted a potential heap buffer overflow vulnerability at the following location in the RT-Thread wlan driver source code:
https://github.com/RT-Thread/rt-thread/blob/master/components/drivers/wlan/wlan_mgnt.c#L215-L226

### Details
Since `len` is a signed integer in the `rt_wlan_send_to_thread()`function, a small negative value could lead to a buffer overflow at the marked lines:
```c
#ifdef RT_WLAN_WORK_THREAD_ENABLE
...
static rt_err_t rt_wlan_send_to_thread(rt_wlan_event_t event, void *buff, int len)
{
    struct rt_wlan_msg *msg;

    RT_WLAN_LOG_D("F:%s is run event:%d", __FUNCTION__, event);

    /* Event packing */
    msg = rt_malloc(sizeof(struct rt_wlan_msg) + len); /* VULN: if len is a small negative number, this would result in an under-allocation */
    if (msg == RT_NULL)
    {
        RT_WLAN_LOG_E("wlan mgnt send msg err! No memory");
        return -RT_ENOMEM;
    }
    rt_memset(msg, 0, sizeof(struct rt_wlan_msg) + len);
    msg->event = event;
    if (len != 0)
    {
        msg->buff = (void *)&msg[1];
        rt_memcpy(msg->buff, buff, len); /* VULN: the small negative number would become a large unsigned size, and we would have a wild memcpy and a heap buffer overflow */
        msg->len = len;
    }

    /* send event to wlan thread */
    if (rt_wlan_workqueue_dowork(rt_wlan_mgnt_work, msg) != RT_EOK)
    {
        rt_free(msg);
        RT_WLAN_LOG_E("wlan mgnt do work fail");
        return -RT_ERROR;
    }
    return RT_EOK;
}
```

### Impact
If the signed size above is confirmed to be attacker-controlled and the input is crossing a security boundary, the impact of the reported buffer overflow vulnerability could range from denial of service to arbitrary code execution.

0xdea, 2023-12-24:
Hi, it's been one month since I reported this vulnerability, and I wanted to ask if you have any update. As standard practice, I plan to request a CVE ID for every confirmed vulnerability. I also intend to publish an advisory by February at the latest, unless there's a specific reason to postpone. Thanks!

cms19859230182-lang, 2026-11-10:
/claim

closed 2026-11-12 by Rbb666