RT-Thread/rt-thread #8289: Potential static buffer overflow in RT-Thread rt-link utility

0xdea, 2023-11-24:
Hi,

I would like to report another potential vulnerability in the current version of RT-Thread. Please let me know if you plan to ask for a CVE ID in case the vulnerability is confirmed. I'm available if you need further clarifications.

## Potential static buffer overflow in RT-Thread rt-link utility

### Summary
I spotted a potential static buffer overflow vulnerability at the following location in the RT-Thread rt-link utility source code:
https://github.com/RT-Thread/rt-thread/blob/master/components/utilities/rt-link/src/rtlink.c#L239

### Details
Lack of length check in the `frame_send()` function could lead to a static buffer overflow at the marked line:
```c
static rt_ssize_t frame_send(struct rt_link_frame *frame)
{
    rt_size_t length = 0;
    rt_uint8_t *data = RT_NULL;

    rt_memset(rt_link_scb->sendbuffer, 0, sizeof(rt_link_scb->sendbuffer));
    data = rt_link_scb->sendbuffer;
    length = RT_LINK_HEAD_LENGTH;
    if (frame->head.crc)
    {
        length += RT_LINK_CRC_LENGTH;
    }
    if (frame->head.extend)
    {
        length += RT_LINK_EXTEND_LENGTH;
    }

    length += frame->data_len;
    frame->head.length = frame->data_len;
    rt_memcpy(data, &frame->head, RT_LINK_HEAD_LENGTH);
    data = data + RT_LINK_HEAD_LENGTH;
    if (frame->head.extend)
    {
        rt_memcpy(data, &frame->extend, RT_LINK_EXTEND_LENGTH);
        data = data + RT_LINK_EXTEND_LENGTH;
    }
    if (frame->attribute == RT_LINK_SHORT_DATA_FRAME || frame->attribute == RT_LINK_LONG_DATA_FRAME)
    {
        rt_memcpy(data, frame->real_data, frame->data_len); /* VULN: static buffer overflow, if frame->data_len > 1024 - 4 (it's a rt_uint16_t so at least in theory can be up to 65535) */
        data = data + frame->data_len;
    }
    if (frame->head.crc)
    {
        frame->crc = rt_link_scb->calculate_crc(RT_FALSE, rt_link_scb->sendbuffer, length - RT_LINK_CRC_LENGTH);
        rt_memcpy(data, &frame->crc, RT_LINK_CRC_LENGTH);
    }

    LOG_D("frame send seq(%d) len(%d) attr:(%d), crc:(0x%08x).", frame->head.sequence, length, frame->attribute, frame->crc);
    return rt_link_hw_send(rt_link_scb->sendbuffer, length);
}
```

### Impact
If the unchecked input above is confirmed to be attacker-controlled and crossing a security boundary, the impact of the reported buffer overflow vulnerability could range from denial of service to arbitrary code execution.

0xdea, 2023-12-24:
Hi, it's been one month since I reported this vulnerability, and I wanted to ask if you have any update. As standard practice, I plan to request a CVE ID for every confirmed vulnerability. I also intend to publish an advisory by February at the latest, unless there's a specific reason to postpone. Thanks!

0xdea, 2024-02-08:
Hi there, CVE-2024-25395 was assigned to this vulnerability. I'm planning to publish my security advisory and writeup on March 5th. Thanks.

cms19859230182-lang, 2026-11-10:
/claim

closed 2026-11-12 by Rbb666