RT-Thread/rt-thread #8292: Use of outdated lwIP version in RT-Thread

0xdea, 2023-11-24:
Hi again,

I noticed that the lwIP dependency is outdated: lwIP 1.4.1, 2.0.3, and 2.1.2 are distributed with RT-Thread, while the latest stable version according to https://savannah.nongnu.org/projects/lwip/ is 2.2.0.

This new version, for instance, fixes the following bugs that are present in:
https://github.com/RT-Thread/rt-thread/blob/master/components/net/lwip/lwip-2.0.3/src/apps/httpd/makefsdata/makefsdata.c
https://github.com/RT-Thread/rt-thread/blob/master/components/net/lwip/lwip-2.1.2/src/apps/http/makefsdata/makefsdata.c

```c
int process_file(FILE *data_file, FILE *struct_file, const char *filename)
{
  char varname[MAX_PATH_LEN];
  int i = 0;
  char qualifiedName[MAX_PATH_LEN];
  int file_size;
  u16_t http_hdr_chksum = 0;
  u16_t http_hdr_len = 0;
  int chksum_count = 0;
  u8_t flags = 0;
  const char* flags_str;
  u8_t has_content_len;
  u8_t* file_data;
  int is_compressed = 0;

  /* create qualified name (@todo: prepend slash or not?) */
  sprintf(qualifiedName,"%s/%s", curSubdir, filename); /* BUG, unbounded sprintf() */
  /* create C variable name */
  strcpy(varname, qualifiedName); /* BUG, see above */
  /* convert slashes & dots to underscores */
  fix_filename_for_c(varname, MAX_PATH_LEN);
  register_filename(varname);
...
  strcpy(lastFileVar, varname); /* BUG, see above */

  /* write actual file contents */
  i = 0;
  fprintf(data_file, NEWLINE "/* raw file data (%d bytes) */" NEWLINE, file_size);
  process_file_data(data_file, file_data, file_size);
  fprintf(data_file, "};" NEWLINE NEWLINE);
  free(file_data);
  return 0;
}
```

In my opinion this is not necessarily an actual security concern, because any potential vulnerabilities I spotted are located in the makefsdata utility source code that is unlikely to be run with untrusted input in scenarios where it might matter. That said, other more important bugs might get fixed upstream.

BernardXiong, 2023-11-25:
Due to compatibility reasons, the old version is still retained. Especially for the version lwIP 1.4.1.

And the makefsdata function and most of apps under `lwip-xxx/src/apps` is not used in RT-Thread.

0xdea, 2023-11-26:
Thank you for the clarification. At the moment I don't have any relevant lwIP vulnerabilities to report, so I think this issue can be safely closed.

BernardXiong, 2023-11-28:
Thank you for your feedback.

closed 2023-11-28 by BernardXiong

0xdea, 2023-12-05:
Hi, I just wanted to let you know about these vulnerabilities in lwIP and TinyDir that I've recently reported and are now fixed:

https://git.savannah.nongnu.org/cgit/lwip.git/commit/?id=ee1523630a81fffa6b7d93dd0c7a6191de5856cd
https://git.savannah.nongnu.org/cgit/lwip.git/commit/?id=b413b040936f48d4cd9ed632ac579542c710efae
https://github.com/hnsecurity/vulns/blob/main/HNS-2023-04-tinydir.txt