RT-Thread/rt-thread #8296: Cannot build rt-thread for raspberry pi 4 OS 32 bit

DerrickLau, 2023-11-25:
> scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build\applications\main.o
In file included from C:\Derrick\rt-thread\20231011_rt-thread-master\rt-thread-master\components\drivers\include/rtdevice.h:82,
                 from applications\main.c:12:
C:\Derrick\rt-thread\20231011_rt-thread-master\rt-thread-master\components\drivers\include/drivers/mmcsd_core.h:253:61: error: unknown type name 'uint8_t'; did you mean 'wint_t'?
  253 | rt_int32_t read_lba(struct rt_mmcsd_card *card, size_t lba, uint8_t *buffer, size_t count);
      |                                                             ^~~~~~~
      |                                                             wint_t
scons: *** [build\applications\main.o] Error 1
scons: building terminated because of errors.

I'm building on Windows 10.  Using arm-elf-toolchain-10.3.

BernardXiong, 2023-11-29:
Could you try to update rtconfig.h with menuconfig? I have tried it on Linux and it is OK.

#8311

DerrickLau, 2023-12-03:
Sorry...I just saw the link to the issue number in your comment.  Did you want me to download the latest source code first?

BernardXiong, 2023-12-05:
> Sorry...I just saw the link to the issue number in your comment. Did you want me to download the latest source code first?

Yes, you can try it with the latest version. And it's better to gradually open more features to try it.

closed 2025-02-07 by supperthomas