RT-Thread/rt-thread #8325: ch32_drivers: rt_hw_i2c_init 崩溃

xfan1024, 2023-12-03:
`rt_hw_i2c_init`使用了pin api，但是`rt_hw_i2c_init`可能先于`rt_hw_pin_init`调用

在我的编译结果遇到了这个崩溃

```
                0x00000000000085b0                __rt_init_start = .
 *(SORT_BY_NAME(.rti_fn*))
 .rti_fn.0      0x00000000000085b0        0x4 build/kernel/src/components.o
                0x00000000000085b0                __rt_init_rti_start
 .rti_fn.0.end  0x00000000000085b4        0x4 build/kernel/src/components.o
                0x00000000000085b4                __rt_init_rti_board_start
 .rti_fn.1      0x00000000000085b8        0x4 build/libraries/hal_drivers/drv_soft_i2c.o
                0x00000000000085b8                __rt_init_rt_hw_i2c_init
 .rti_fn.1      0x00000000000085bc        0x4 build/libraries/hal_drivers/drv_gpio.o
                0x00000000000085bc                __rt_init_rt_hw_pin_init
 .rti_fn.1.end  0x00000000000085c0        0x4 build/kernel/src/components.o
                0x00000000000085c0                __rt_init_rti_board_end
 .rti_fn.4      0x00000000000085c4        0x4 build/kernel/components/drivers/i2c/i2c_core.o
                0x00000000000085c4                __rt_init_rt_i2c_core_init
 .rti_fn.4      0x00000000000085c8        0x4 build/kernel/components/libc/cplusplus/cxx_crt_init.o
                0x00000000000085c8                __rt_init_cplusplus_system_init
 .rti_fn.6      0x00000000000085cc        0x4 build/kernel/components/finsh/shell.o
                0x00000000000085cc                __rt_init_finsh_system_init
 .rti_fn.6.end  0x00000000000085d0        0x4 build/kernel/src/components.o
                0x00000000000085d0                __rt_init_rti_end
                0x00000000000085d4                __rt_init_end = .
```

[崩溃位置](https://github.com/RT-Thread/rt-thread/commit/90d9b449b1c7eb370545244c26e2d92446b42d84#r134068695)

BernardXiong, 2023-12-03:
调整下，或者是否可以做个什么关联，让gpio先行启动。

xfan1024, 2023-12-03:
> 调整下，或者是否可以做个什么关联，让gpio先行启动。

我按我项目里的做法给在master分支提了个PR，麻烦看下是否合理。

PR #8328 (2023-12-16): ch32_drivers: call rt_hw_pin_init in rt_hw_i2c_init explicitly
Fix #8325

## 拉取/合并请求描述：(PR description)

#### 为什么提交这份PR (why to submit this PR)

在Linux下构建项目时，`rt_hw_i2c_init`先于`rt_hw_pin_init`被调用，导致崩溃。

#### 你的解决方案是什么 (what is your solution)

在`rt_hw_i2c_init`中显式调用`rt_hw_pin_init`，确保`pin`模块先于`i2c`模块初始化。

#### 请提供验证的bsp和config (provide the config and bsp) 

- BSP:  在私有的BSP上验证过(version 4.1.1)，master分支未测试

### 当前拉取/合并请求的状态 Inte
closed 2023-12-16 by mysterywolf