RT-Thread/rt-thread #8358: 🎯 rt-smart for riscv64 TODO list

BernardXiong, 2023-12-10:
列出一些rt-smart for riscv64上需要完成的工作事项，希望社区有更多人参与，一起来完成：
* 升级musl libc版本
  * 因为目前主要使用的musl libc版本是musl-1.1.22版本，而musl对riscv的支持是从1.1.23开始。目前最新的musl libc版本是1.2.4；
  * 解决`__riscv_flush_icache`的问题；
* 加入riscv64的ld.so支持，从而支持到动态链接的应用程序；
* backtrace，coredump，gdb stub支持到riscv64架构；

flyingcys, 2023-12-12:
backtrace我之前有看过一些原理，刚好可以实操一下，backtrace任务我领走。
RISCV64的libcpu统一，也可以我来推进。

polarvid, 2024-05-06:
补充一些 TODO

- [ ] 参考 aarch64 PR #8672 支持 arch_syscall_restart

unicornx, 2024-12-04:
> * riscv64 cpu，内存，UART，中断相关初始化都可以支持到设备树；
该问题已经建立 issue #9707 进行跟踪，先在 duo 上实现支持。

unicornx, 2024-12-04:
> * [ ]  统一riscv64的libcpu，把virt，c906等的都统一起来；

该问题已经建立如下 issue 进行跟踪。

- #9743

unicornx, 2024-12-04:
> * [ ]  应用验证：hello world/ Busybox init/ash/gzip（待确认）
> * [ ]  应用验证：vim（待确认）
> * [ ]  应用验证：dropbear（待确认）

@polarvid 请问还能详细描述一下，这个理的应用验证是什么意思？我好对应建立 issue 来 track

unicornx, 2024-12-04:
> ## RVV 1.0 合入主线
> * [ ]  在 c908/virt64 libcpu 移植中已完成支持，后续期望在 common64 中统一支持该 ISA（待确认）

@polarvid 请问这个是否已经实现了? 我看到下面这两个 commit，rvv 1.0 似乎已经统一在 `libcpu/risc-v/vector/rvv-1.0` 中。

```
commit 3268716c4fdd506487f89b39b09b8f5339882ae5
Author: heyuanjie87 <943313837@qq.com>
Date:   Thu Nov 21 17:18:35 2024 +0800

    [libcpu][riscv]virt64使用通用vector支持代码

 libcpu/risc-v/virt64/SConscript        |   4 ++--
 libcpu/risc-v/virt64/rvv_context.h     | 111 ---------------------------------------------------------------------------------------------------------------
 libcpu/risc-v/virt64/vector_encoding.h |  51 ---------------------------------------------------
 libcpu/risc-v/virt64/vector_gcc.S      |  45 ---------------------------------------------
 4 files changed, 2 insertions(+), 209 deletions(-)
```

和 

```
commit 49b661476393bd9c9fc199362cf4b008c35a9340
Author: heyuanjie87 <943313837@qq.com>
Date:   Wed Oct 16 14:10:09 2024 +0800

    [libcpu]添加对riscv vector的支持 (#9531)
    
    [libcpu]添加对riscv vector的支持

 libcpu/Kconfig                                 |  12 ++++++++++++
 libcpu/risc-v/common64/stackframe.h            |   3 ++-
 libcpu/risc-v/t-head/c908/SConscript           |   4 ++--
 libcpu/risc-v/vector/rvv-1.0/rvv_context.h     | 112 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 libcpu/risc-v/vector/rvv-1.0/vector_encoding.h |  55 +++++++++++++++++++++++++++++++++++++++++++++++++++++++
 5 files changed, 183 insertions(+), 3 deletions(-)
```

unicornx, 2024-12-04:
> ## 进程管理
> * [ ]  参考 aarch64 PR [sync smart & dfs #8672](https://github.com/RT-Thread/rt-thread/pull/8672) 支持 arch_syscall_restart
> 

@polarvid 请问具体这个具体是指什么工作？是否已经完成了？

heyuanjie87, 2024-12-04:
rvv已经统一了

zhuzhuzhus, 2024-12-04:
> > * [ ]   应用验证：hello world/ Busybox init/ash/gzip（待确认）
> > * [ ]   应用验证：vim（待确认）
> > * [ ]   应用验证：dropbear（待确认）
> 
> @polarvid 请问还能详细描述一下，这里三个”应用验证“ 具体是什么意思？我好对应建立 issue 来 track

就是确认这几个app在riscv上能不能跑，hello world/ Busybox init/ash/gzip/vim 在userapp中都有，dropbear后续会推出

polarvid, 2024-12-09:
> > * [ ]   统一riscv64的libcpu，把virt，c906等的都统一起来；
> 
> 该问题已经建立如下 issue 进行跟踪。
> 
> * [[Feature] 统一 riscv64 的 libcpu #9743](https://github.com/RT-Thread/rt-thread/issues/9743)
> 
> @polarvid 需要确认一下，#9151 就是针对这个问题的工作吗？这个统一的工作你觉得目前是做完了吗？

是的。

polarvid, 2024-12-09:
> > * [ ]   ARCH_REMAP_KERNEL cvitek bsp 支持（待确认）
> 
> @polarvid 这个是不是对应的 [148e577](https://github.com/RT-Thread/rt-thread/commit/148e5774c96f84c326452ebe2ee7690f2224949a) 的修改？
> 
> ```
> commit 148e5774c96f84c326452ebe2ee7690f2224949a
> Author: Shell <smokewood@qq.com>
> Date:   Fri Jul 26 10:12:13 2024 +0800
> 
>     bsp: enable KERNEL_REMAP for cvitek platform
>     
>     Changes:
>     
>     - board.c: add static assert for KERNEL_VADDR_START to check if it's valid
>     - board.h: updated deafult KERNEL_VADDR_START for standard version
>     - config bsp for v5.2.0 smart requirements
>     - kconfig: update bsp Kconfig for remap kernel
>     
>     Signed-off-by: Shell <smokewood@qq.com>
>     Reviewed-on: https://github.com/RT-Thread/rt-thread/pull/9229
>     Reviewed-by: Chen Wang <unicorn_wang@outlook.com>
> ```

是的。

polarvid, 2024-12-09:
> > ## 进程管理
> > 
> > * [ ]   参考 aarch64 PR [sync smart & dfs #8672](https://github.com/RT-Thread/rt-thread/pull/8672) 支持 arch_syscall_restart
> 
> @polarvid 请问具体这个具体是指什么工作？是否已经完成了？

如题所述，需要提供一个系统调用重启的实现。算法可以参考 aarch64 的 PR 。

unicornx, 2024-12-09:
> > > * [ ]    统一riscv64的libcpu，把virt，c906等的都统一起来；
> > 
> > 
> > 该问题已经建立如下 issue 进行跟踪。
> > 
> > * [[Feature] 统一 riscv64 的 libcpu #9743](https://github.com/RT-Thread/rt-thread/issues/9743)
> > 
> > @polarvid 需要确认一下，#9151 就是针对这个问题的工作吗？这个统一的工作你觉得目前是做完了吗？
> 
> 是的。

ok, #9743 关闭了。

unicornx, 2024-12-09:
> > > * [ ]    应用验证：hello world/ Busybox init/ash/gzip（待确认）
> > > * [ ]    应用验证：vim（待确认）
> > > * [ ]    应用验证：dropbear（待确认）
> > 
> > 
> > @polarvid 请问还能详细描述一下，这里三个”应用验证“ 具体是什么意思？我好对应建立 issue 来 track
> 
> 就是确认这几个app在riscv上能不能跑，hello world/ Busybox init/ash/gzip/vim 在userapp中都有，dropbear后续会推出

先提了 #9760 来耿总

unicornx, 2024-12-09:
> > > ## 进程管理
> > > 
> > > * [ ]    参考 aarch64 PR [sync smart & dfs #8672](https://github.com/RT-Thread/rt-thread/pull/8672) 支持 arch_syscall_restart
> > 
> > 
> > @polarvid 请问具体这个具体是指什么工作？是否已经完成了？
> 
> 如题所述，需要提供一个系统调用重启的实现。算法可以参考 aarch64 的 PR 。

提了  #9761 来 track

unicornx, 2024-12-09:
@BernardXiong 这个 issue 上列的问题，已经全部确认完毕，有些已经完成，未做的也都分别提了 issue 单独 track 了，这个 issue 可以关闭了。

BernardXiong, 2024-12-17:
> @BernardXiong 这个 issue 上列的问题，已经全部确认完毕，有些已经完成，未做的也都分别提了 issue 单独 track 了，这个 issue 可以关闭了。

🆗Thanks

closed 2024-12-17 by BernardXiong