RT-Thread/rt-thread #8420: milk-v duo开发板移植RT-Thread Smart出现'struct rt_device' has no member named 'ops'; did you mean 'fops'?报错

2021zhangyixu, 2023-12-26:
昨天我朋友尝试用duo移植RT-Thread Smart，后面他说出现了报错。我跟着尝试了一下，发现确实存在这个问题。之后我修改了components/dfs/dfs_v2/filesystems/devfs/devfs.c文件中的285行，315行，347行，378行，420行的ops为fops。最终编译通过了，上机测试也没有问题。
[image]
可能是某位大佬写的时候不小心将ops和fops弄混了。
[image]

BernardXiong, 2023-12-26:
@geniusgogo 是否和某个PR相关？

polarvid, 2024-01-05:
SMART 现在强制选用 DFS 2.0 了。然后 dfs 2.0 的 devfs 只支持 fops 形式的 RT device 原型。所以有些旧的驱动要主动更新一下。risc-v 的 bsp 就有很多这个问题。

tang5722917, 2024-03-22:
这个bug应该是bsp缺少定义#define RT_USING_DEVICE_OPS导致的
以285行error为例
```
//devfs.c 285行
#ifdef RT_USING_POSIX_DEVIO
    if (device->fops && device->fops->ioctl)
    {
        result = device->fops->ioctl(file, cmd, args);
    }
    else if (device->ops)
#else
```
此处的逻辑为判断device是否为fops 文件设备，如果不是就是普通IO设备ops
依据 device 的定义为 rt_device_t device;
其中rt_device_t 为结构体rt_device 指针
```
//rtdef.h  rt_device定义  1314行
struct rt_device
{
    struct rt_object          parent;                   /**< inherit from rt_object */

#ifdef RT_USING_DM
    struct rt_bus *bus;                                 /**< the bus mounting to */
    rt_list_t node;                                     /**< to mount on bus */
    struct rt_driver *drv;                              /**< driver for powering the device */
#ifdef RT_USING_OFW
    void *ofw_node;                                     /**< ofw node get from device tree */
#endif /* RT_USING_OFW */
#endif /* RT_USING_DM */

    enum rt_device_class_type type;                     /**< device type */
    rt_uint16_t               flag;                     /**< device flag */
    rt_uint16_t               open_flag;                /**< device open flag */

    rt_uint8_t                ref_count;                /**< reference count */
    rt_uint8_t                device_id;                /**< 0 - 255 */

    /* device call back */
    rt_err_t (*rx_indicate)(rt_device_t dev, rt_size_t size);
    rt_err_t (*tx_complete)(rt_device_t dev, void *buffer);

#ifdef RT_USING_DEVICE_OPS
    const struct rt_device_ops *ops;
#else
    /* common device interface */
    rt_err_t  (*init)   (rt_device_t dev);
    rt_err_t  (*open)   (rt_device_t dev, rt_uint16_t oflag);
    rt_err_t  (*close)  (rt_device_t dev);
    rt_ssize_t (*read)  (rt_device_t dev, rt_off_t pos, void *buffer, rt_size_t size);
    rt_ssize_t (*write) (rt_device_t dev, rt_off_t pos, const void *buffer, rt_size_t size);
    rt_err_t  (*control)(rt_device_t dev, int cmd, void *args);
#endif /* RT_USING_DEVICE_OPS */

#ifdef RT_USING_POSIX_DEVIO
    const struct rt_device_t *fops;
    struct rt_wqueue wait_queue;
#endif /* RT_USING_POSIX_DEVIO */

    void                     *user_data;                /**< device private data */
};
```
只要添加宏定义RT_USING_DEVICE_OPS就能让device包含ops变量

你把ops改为fops 不符合if-else的逻辑

tang5722917, 2024-03-22:
我也遇到了相同的问题，使用全志d1s的bsp，同样是将报错的ops改为fops

closed 2024-03-25 by mysterywolf