RT-Thread/rt-thread #8427: [ci] 继续完善编译strick mode

mysterywolf, 2023-12-26:
配合attachconfig文件对指定bsp进行严格编译

scons --strict="-Werror -Wunsed -Wextra"

.attach在附加宏之前可以标定本次编译的严格程度
```
# strict: -Werror -Wunsed -Wextra
```

mysterywolf, 2023-12-28:
attachconfig 文件夹支持递归扫描子文件夹下的attach文件

vacabun, 2023-12-28:
我个人也比较倾向于方案二，在每个.attach文件的开头添加：
`# scons: --strict="-Werror -Wunsed -Wextra"`
或
`# --strict="-Werror -Wunsed -Wextra"`

mysterywolf, 2023-12-30:
经过思考，不同的组采用不同的严格程度是更符合rt-thread实际情况的，scons --strict保持现状，无需增加--strict=“”方式，具体参数添加有Group确定 详见：https://github.com/RT-Thread/rt-thread/pull/8454

.attach文件仅需完成如下解析即可

```
# scons: --strict
```

如果侦测到scons关键字，则编译时，补充使用冒号后面的参数

vacabun, 2024-01-04:
https://github.com/RT-Thread/rt-thread/pull/8464

检测到`# scons: [scons_args]`，编译时会补充使用冒号后面的参数，即添加到`scons -C bsp/{bsp} -j{nproc} {scons_args}`。

`.attach`文件内可以有多行`# scons: [scons_args]`，会自动以空格分开。

closed 2024-01-09 by mysterywolf