RT-Thread/rt-thread #8428: stm32串口掉数据

cc-caixf, 2023-12-27:
场景: 
- 串口1 `115200bps` 以`中断`的方式接收/发送数据
- 串口2 `115200bps` 以`DMA`的方式接收/发送数据

现象
- 串口2接收大批量数据时, 串口1就会出现接收的数据包中掉1个字节的情况

分析
- `serial_v2.c` 的[这里](https://github.com/RT-Thread/rt-thread/blob/ed4d037d7b09cee8e53810affd215ae504c79fc7/components/drivers/serial/serial_v2.c#L432C16-L432C16)从ringbuf copy数据长时间的关闭中断, 导致系统响应中断不及时. 将这里的关中断去掉, 串口1就不会掉数据(话说, 这里的互斥不应该应用层来做吗?)

还发现的问题
- `drv_usart.c` 的stm32 实现, ORE在读SR再读DR就会被清掉, 所以[这种写法](https://github.com/RT-Thread/rt-thread/blob/ed4d037d7b09cee8e53810affd215ae504c79fc7/bsp/stm32/libraries/HAL_Drivers/drivers/drv_usart.c#L461C16-L461C16)根本不能发现ORE, 判断ORE应该在读DR之前

chnykn, 2023-12-29:
你提到的“1个字节”，是不是在遇到 \r\n 两个字符的时候，\n没有了？

如果是这种情况的话，原因在于：
	./rt-thread/components/net/at/src/at_client.c 的at_recv_readline和client_parser函数中
	如果传输的是二进制文件，其中内容当中凑巧有\r\n的字符，at_recv_readline函数会解释为换行
	随后client_parser函数当中，会把把\n替换为\0, 导致内容变化。

cc-caixf, 2023-12-29:
> 你提到的“数据包中掉**1个**字节”，是不是在遇到 \r\n 两个字符的时候，\n没有了？
> 
> 如果是这种情况的话，原因在于： ./rt-thread/components/net/at/src/at_client.c 的at_recv_readline和client_parser函数中 如果传输的是二进制文件或数据，其中内容当中凑巧有\r\n的字符，at_recv_readline函数会解释为换行 随后client_parser函数当中，会把把\n替换为\0, 导致内容变化。

我是串口接收数据流, 没用到at, 跟你描述的不是同一个现象/问题.

thewon86, 2024-01-04:
https://club.rt-thread.org/ask/article/bfd92159ba11aef6.html
这里的开关中断换成只开关当前串口外设的中断就不会影响到其它串口了。

cc-caixf, 2024-01-12:
这个串口DMA接收实现就是个大坑!!!!!
现在的逻辑是: DMA一直接收, 根本不管buffer有没有溢出, 这将导致: 如果buffer溢出(buffer里面的值已经不知道重复修改了几次, len值不能反应buffer内的数据量), 上层(应用层)根本感知不到, 上层怎么处理都恢复不了正常状态, 只有重启.
正常的逻辑不应该是: DMA根据buffer的余量接收, 如果buffer满了, 就不应该接收, 等上层处理完了(buffer有余量了), 在接收数据啊.

thewon86, 2024-01-15:
> 这个串口DMA接收实现(stm32)就是个大坑!!!!!
> 
> 现在的逻辑是: DMA一直接收, 根本不管buffer有没有溢出, 这将导致: 如果buffer溢出(buffer里面的值可能已经修改了几伦, 然而len值已经不能反应buffer内的真实数据量), 上层(应用层)根本感知不到buffer已经溢出(buffer中的数据已经覆盖再覆盖了, len已经背离真实情况), 上层怎么处理都恢复不了正常状态, 除非关闭再打开串口(清除底层buffer的len, 让len恢复正常).
> 
> 正常的逻辑不应该是: DMA根据buffer的余量接收, 如果buffer满了, 就不应该接收, 等上层处理完了(buffer有余量了), 在接收数据啊.

还可以保证新的数据全部就收，丢弃最旧的数据，反正应用层已经处理不过来了，总需要丢掉一些数据。

放弃 v2 吧，论坛上还有个人多read了数据的，提问了不止三次了，没人敢解答，v2 的问题没人帮你解决得了的

cc-caixf, 2024-01-15:
> 还可以保证新的数据全部就收，丢弃最旧的数据，反正应用层已经处理不过来了，总需要丢掉一些数据。
>
> 放弃 v2 吧，论坛上还有个人多read了数据的，提问了不止三次了，没人敢解答，v2 的问题没人帮你解决得了的

看了博流的串口DMA RX实现, 感觉这种实现方式很合理, 先DMA到指定`底层buffer`中, 再在半满/全满/idle中断, 将`底层buffer`中的数据读到`ringbuffer`中, 应用层再从`ringbuffer`中读数据. 这样ringbuffer满不满(应用层处理不处理的过来)跟DMA无关

closed 2025-09-28 by Rbb666