RT-Thread/rt-thread #8459: ready list插入顺序问题

daguobayern, 2024-01-02:
版本：master分支最新commit
背景：结合两次修复历史（https://github.com/RT-Thread/rt-thread/pull/6232与https://github.com/RT-Thread/rt-thread/issues/8050）
都是针对ready list add 成员时做出的修复，其中rt_schedule_insert_thread函数中代码如下：
    /* there is no time slices left(YIELD), inserting thread before ready list*/
    if((thread->stat & RT_THREAD_STAT_YIELD_MASK) != 0)
    {
        rt_list_insert_before(&(rt_thread_priority_table[thread->current_priority]),
                              &(thread->tlist));
    }
    /* there are some time slices left, inserting thread after ready list to schedule it firstly at next time*/
    else
    {
        rt_list_insert_after(&(rt_thread_priority_table[thread->current_priority]),
                              &(thread->tlist));
    }
我们的本意应该是如果时间片没用完被调度切换到更高优先级的线程时，那么在下次切回时应优先执行未使用完时间片的线程，例如A，B，C三线程，优先级值为A=B>C，在A执行时，C由于优先级最高被调度执行（假设被中断触发信号量），那么在C执行完时，应该先执行A，待A时间片轮完再执行B，以此交替。总结起来就是同优先级的线程必须按顺序依次执行完各自的时间片，中途可被中断或更高优先级打断，但总顺序不变，这样可以保证同优先级线程最大的公平性。

问题1：
假设线程ABCD，D优先级最高，ABC优先级相同，线程A，B，C依次获取信号量s从而挂起（s的flag为fifo），那么在插入s的挂起线程链表时调用rt_list_insert_before函数，其挂起线程链表中顺序为A->B->C，在某个时刻释放信号量时，依次调用函数_ipc_list_resume，rt_thread_resume，rt_schedule_insert_thread，此时将A取出插入相应优先级的ready list中，由于没有RT_THREAD_STAT_YIELD，那么是插队插入ready链表头中，调度时由于优先级低D继续执行，而后又释放了一次信号量，相同的步骤将B插队插入ready list中，此时ready list中链表顺序为B->A，若D此时释放CPU，那么调度先执行B而不是A，这显然违背了我们的意图。

修复：
插队插入链表头只能是时间片未用完的情况，其余大部分情况应该都是顺序插入链表尾，包括线程的主动挂起（rt_thread_yield）。

zhzhqian, 2024-01-04:
>如A，B，C三线程，优先级值为A=B>C，在A执行时，C由于优先级最高被调度执行（假设被中断触发信号量），那么在C执行完时，应该先执行A，待A时间片轮完再执行B
>而后又释放了一次信号量，相同的步骤将B插队插入ready list中，此时ready list中链表顺序为B->A，若D此时释放CPU，那么调度先执行B而不是A
这里没太理解，假如有两个同优先级的线程A，B时间片均未用完，A，B先后让出CPU（阻塞在锁或者信号量），那么当高优先级线程先后唤醒A，B（或者同时唤醒）并让出CPU，应该先执行哪一个线程呢？如果是最近被让出的线程，也就是B，那么我理解这个插入ready list的逻辑就不存在问题，如果是A，那么我想问为什么？选择A而不是B对系统公平性或者性能有什么正向影响吗？

daguobayern, 2024-01-04:
感谢回复，是这个意思，如果A，B是先后获取信号量而挂起的，而这个信号量的flag是RT_IPC_FLAG_PRIO，表示在有信号量时按挂起顺序来执行线程，即先执行A后再执行B。
在高优先级线程C中释放信号量时，rt_sem_release/_ipc_list_resume函数取出挂起的线程插入到ready list中，由于是先A后B获取的信号量，挂起线程中的链表顺序是A->B，那么第一次取出的线程是A，插入到ready list的是使用rt_list_insert_after，即插入到头部。而后发生调度，此时执行D，D再次释放信号量，同样的流程，这次从挂起线程链表中取出的线程是B，插入到ready list的也是使用rt_list_insert_after，那么会插入到A的前面，即此时ready list的链表顺序是B->A，再次调度时则会先执行B再执行A，这就违背FIFO属性了。

wb2712, 2025-05-06:
这么核心的功能，还没修复好？

Yunlk, 2026-10-20:
## 主机侧最小复现：同优先级线程按信号量顺序唤醒时，ready list 会逆序

版本：master `1c801efa`（2026-10-19）

先给结论：`src/scheduler_up.c` 里 `rt_schedule_insert_thread` 的 ready list 插入段（现为 507-520 行），与报告里贴出的代码**逐字一致**，当前 master 仍是这段代码。

我把这段插入逻辑单独拆出来跑了一遍，能稳定复现逆序。

### 机制

`include/rtservice.h`：

- `rt_list_insert_after(&head, n)` → n 落在**队首**
- `rt_list_insert_before(&head, n)` → n 落在**队尾**

`include/rtdef.h:638,640`：`RT_THREAD_STAT_MASK = 0x07`，`RT_THREAD_STAT_YIELD = 0x08`。YIELD 位不在 STAT_MASK 内，所以第 508 行的 `stat & ~RT_THREAD_STAT_MASK` 不会误清它，分支判断本身是有效的。

但正因如此：**由 IPC（信号量）释放唤醒的线程从未调用过 `rt_thread_yield()`，YIELD 位为 0，因此全部走 else 分支，被插到队首。** 后唤醒的线程排到最前面，先唤醒的被顶到后面。

### 实测

直接复用 `rtservice.h:61/76` 与 `scheduler_up.c:507-520` 的原始实现，逻辑未做任何改动：

```
[1] A、B 挂起在 flag=PRIO 的信号量上，pending 顺序 A->B，
    先后两次 release 唤醒
    ready queue 实际: B A         期望: A B

[2] 同上，A、B、C 三个线程
    ready queue 实际: C B A       期望: A B C   （完全逆序）

[3] 同样的 A、B，但两者 YIELD 位均置位
    ready queue 实际: A B         （顺序正常）
```

`[1]` 与 `[3]` 的差异正好落在 else 分支上，问题位置可以唯一确定。

说明：这是**主机侧的链表顺序模拟**，只验证 ready list 最终排列，不是真机/BSP 上的调度实测。它证明的是插入结果与报告预期不符，不能替代设备侧观测。

### 想确认两点（不是催结论）

1. `[3]` 与 `[1]` 的行为差异是否符合设计意图？也就是说，未 yield 过的线程是否**本就应当**优先于同优先级中已排队的线程？

2. 若确认是缺陷，维护者更倾向哪种方向：把 `RT_THREAD_STAT_YIELD` 的语义收窄为"仅时间片未用尽被抢占"、其余唤醒路径（IPC 恢复等）一律走队尾插入；还是维持现状、把同优先级 FIFO 保证的范围写进文档？

需要的话我可以补一份 qemu-vexpress-a9 上的设备侧调度顺序测试，把主机侧的推断落到真机日志上。

Yunlk, 2026-10-20:
## 补充：qemu-vexpress-a9 设备侧实测（含因果验证）

上一条只做了主机侧链表模拟，补一份真机侧数据。

环境：master `1c801efa`，`bsp/qemu-vexpress-a9`，QEMU vexpress-a9，RT-Thread 5.4.0，arm-none-eabi-gcc 15.2.1。

方法：同优先级线程（priority 20），T1/T2 全程不调用 `rt_thread_yield()`，用串口日志观察实际运行顺序。

### T1 启动顺序

先 `rt_thread_startup(A)`，紧接着 `rt_thread_startup(B)`，中间不让出 CPU：

```
[B] pend on sem
[A] pend on sem
```

A 先就绪，B 却先运行。此时 ready list 已经是 B->A。

### T2 信号量唤醒（pending 顺序强制为 A->B）

A 先启动并阻塞在信号量上，再启动 B，保证 pending 链表确定是 A->B：

```
[A] pend on sem
[B] pend on sem
--- release #1 (wakes A) ---
--- release #2 (wakes B) ---
[B] resumed, running now
[A] resumed, running now
```

实际执行顺序 **BA**，与预期 A->B 相反。`RT_IPC_FLAG_PRIO` 与 `RT_IPC_FLAG_FIFO` 两种 flag 结果一致。

### T3 线程先 yield 再阻塞

两个线程各调用一次 `rt_thread_yield()` 后再阻塞，结果仍是 **BA**。

原因在 `src/scheduler_up.c:338-341`：

```c
if ((RT_SCHED_CTX(from_thread).stat & RT_THREAD_STAT_YIELD_MASK) != 0)
{
    RT_SCHED_CTX(from_thread).stat &= ~RT_THREAD_STAT_YIELD_MASK;
}
```

线程 yield 之后被切走时 YIELD 位就被清掉了，等它再阻塞、被 IPC 唤醒时该位已是 0，照样走 else 分支插到队首。

顺带修正上一条评论里的一处说法：`insert_before`（队尾）分支在实践中接近不可达。现存代码里，能让 YIELD 位一直保留到 `rt_schedule_insert_thread` 的，基本只剩「时间片未用尽被抢占后立即重新入队」这一条路径；而 IPC 唤醒路径一律被插队。

### 因果验证

把 else 分支临时改为 `rt_list_insert_before`（队尾插入），其余不动，同一套测试重跑：

```
T1: [A] pend on sem 先于 [B] pend on sem
T2: observed run order = AB  -> order PRESERVED
T3: observed run order = AB  -> order PRESERVED
```

反向验证即证明问题确实落在 `scheduler_up.c:515-520` 的 else 分支。

需要说明：这个改动**只用于定位根因，不是修复提案**。队首插入在「时间片未用尽被抢占」的场景下有存在理由（关联 #6232），直接把该分支改成队尾可能引出另外的公平性变化，需要维护者判断。

复现用的是一份约 40 行的应用层测试（`rt_sem_create` + 同优先级线程 + 顺序打印），没有改内核。如果希望进 CI，我可以整理成 utest 用例提交。

Yunlk, 2026-10-20:
/claim