RT-Thread/rt-thread #8490: psoc6-evaluationkit-062S2 set RTC time failed

hywing, 2024-01-14:
Environment

-------------------------------------------------------------

board : psoc6-evaluationkit-062S2 
bsp : PSOC62-IFX-EVAL-KIT 1.1.1
rt-thread : 5.0.1

-------------------------------------------------------------

Here is my code

-------------------------------------------------------------

#include <rtthread.h>
#include <rtdevice.h>
#include <time.h>
#include "drv_gpio.h"
#include "drivers/alarm.h"

#define LED_PIN     GET_PIN(0, 1)

#define DBG_LEVEL DBG_LOG
#include <rtdbg.h>

rt_sem_t rtc_init_sem = RT_NULL;

static int rtc_sample(int argc, char *argv[])
{
     rt_err_t ret = RT_EOK;
     time_t now;
     /* 设置日期 */
     ret = set_date(2024, 1, 14);
     if (ret != RT_EOK)
     {
     rt_kprintf("set RTC date failed\n");
     return ret;
     }
     /* 设置时间 */
     ret = set_time(10, 10, 50);
     if (ret != RT_EOK)
     {
     rt_kprintf("set RTC time failed\n");
     return ret;
     }
     /* 延时3秒 */
     rt_thread_mdelay(3000);
     /* 获取时间 */
     now = time(RT_NULL);
     rt_kprintf("%s\n", ctime(&now));
     return ret;
}
/* 导出到 msh 命令列表中 */
MSH_CMD_EXPORT(rtc_sample, rtc sample);

int main(void)
{
    rt_pin_mode(LED_PIN, PIN_MODE_OUTPUT);
    for (;;)
    {
        rt_pin_write(LED_PIN, PIN_HIGH);
        rt_thread_mdelay(500);
        rt_pin_write(LED_PIN, PIN_LOW);
        rt_thread_mdelay(500);
    }
}

-------------------------------------------------------------

My RT-Thread Studio Settings 

-------------------------------------------------------------
[image]

-------------------------------------------------------------

I get an error like this
[image]

Rbb666, 2024-01-14:
试试这么修改后是否正常？
```
#ifdef   BSP_RTC_USING_LSE
    Cy_RTC_SelectClockSource(CY_RTC_CLK_SELECT_WCO);
#else
    Cy_RTC_SelectClockSource(CY_RTC_CLK_SELECT_ILO);
#endif
```
[image]

hywing, 2024-01-23:
加上了就可以了

Rbb666, 2024-01-24:
> 加上了就可以了

好的，稍后我提PR修复下

closed 2024-01-24 by mysterywolf