RT-Thread/rt-thread #8493: SMP下中断与互斥锁同时使用的存在死锁的场景

Yaochenger, 2024-01-16:
- [    问题      ] SMP模式下中断与互斥锁同时使用的存在死锁，_cpus_lock死锁

- [  测试版本 ] 主线最新版本

- [  测试BSP  ]  qemu-vexpress-a9

- [  测试用例 ]  
```
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <rtthread.h>
#include <rthw.h>
#define NUM 1000000
#define TASK_NUM 3

static volatile unsigned long long count = 0;
static int ac = 0;
static rt_mutex_t test_mutex;

static void test_thread(void *param)
{
    for (int i = 0; i < NUM; i++) {
        rt_mutex_take(test_mutex, RT_WAITING_FOREVER);
        ++count;
        rt_mutex_release(test_mutex);
    }
    printf("count: %llu\n", count);
}

int main(void)
{
    rt_thread_t threads[TASK_NUM];
    test_mutex = rt_mutex_create("test_mutex", RT_IPC_FLAG_PRIO);

    for (int i = 0; i < TASK_NUM; i++) {
        char buf[64];
        rt_snprintf(buf, sizeof(buf), "test%d", i);
        threads[i] = rt_thread_create(buf, test_thread, 0, 2048, 20+i, 10u);
        rt_thread_startup(threads[i]);
    }

    while(1) {
        if (count == (NUM * TASK_NUM)) {
            printf("finish test...\n");
            rt_thread_mdelay(5000);
            continue;
        }
        if (++ac % 5 == 0) {
            printf("count: %llu\n", count);
        }
        rt_thread_mdelay(100);
    }
    return 0;
}
```

- [ 临时解决办法 ] 
以下是目前探索找到的临时的解决办法：
-等待一定时间获取不到锁主动释放
-在互斥锁获取与释放之间添加rt_thread_mdelay

Yaochenger, 2024-01-16:
@BernardXiong

polarvid, 2024-01-17:
看上去更像是调度器的冲突区保护存在问题。

closed 2024-02-23 by BernardXiong