RT-Thread/rt-thread #8494: 关于编译时显示占用的问题

wirano, 2024-01-16:
观察到 scon 在生成 cmake 工程时使用 size 工具显示工程的占用空间。除了 size 工具外，还可以添加 `--gc-sections,--print-memory-usage` 选项让 ld 输出占用信息。相比之下使用 `--gc-sections,--print-memory-usage` 输出的信息更加直观。

请问使用 size 而不是 `--gc-sections,--print-memory-usage` 是有什么考虑吗（`--gc-sections,--print-memory-usage` 依赖 bsp 作者添加链接选项，但这应该可以通过脚本在生成工程阶段实现）

```
   text	   data	    bss	    dec	    hex	filename
  24363	    340	   1344	  26047	   65bf	rtthread.elf
```

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       24364 B       256 KB      9.29%
            SRAM:        1684 B        32 KB      5.14%
```

BernardXiong, 2024-01-22:
支持PR，完善更多信息显示输出

runsunlg, 2024-01-23:
链接选项--gc-sections一般要与编译选项-function-sections,-fdata-sections使用，因此还需要修改编译选项，这几个选项一般是作为优化可执行文件大小用的，去掉了不用的函数和全局变量。

PR #8697 (2024-03-30): [gcc] add footprint information when compiling
## 拉取/合并请求描述：(PR description)

close: #8494

[image]

[

#### 为什么提交这份PR (why to submit this PR)

#### 你的解决方案是什么 (what is your solution)

#### 请提供验证的bsp和config (provide the config and bsp) 

- BSP:

- .config:

- action:

]
closed 2024-03-30 by mysterywolf