RT-Thread/rt-thread #8563: AARCH64，SMP模式下的建议

messigogogo, 2024-02-28:
[image]
在libcpu/aarch64/common/gicv3.c 文件中，arm_gic_redist_init函数用于获取当前CPU的gic地址，并进行相关初始化操作。其中redist_base += cpu_id * (2 << 16); 这一行代码个人感觉不太灵活，因为某些CPU不一定按照0123...的顺序启动，可能最先启动的是2核，所以需要在适配时rt_hw_cpu_id()函数中对cpu_id做繁琐的转换操作。并且在BSP中已经对platform_get_gic_redist_base(void)函数进行了适配，请问是否可以改为redist_base =platform_get_gic_redist_bas()。

messigogogo, 2024-02-28:
这样的话，操作系统认为的cpu的顺序是0，1，2，3...，但实际开发板cpu启动的顺序可以是2，0，1，3...。操作系统理论上只知道每个核gic的基地址就可以，没有必要自己计算，应该不会产生冲突？

BernardXiong, 2024-02-28:
可以提份PR上来吗？或者说，这个cpu_id应该是一个逻辑的id，而不见得是实际id。

messigogogo, 2024-02-29:
好的

messigogogo, 2024-03-04:
> 可以提份PR上来吗？或者说，这个cpu_id应该是一个逻辑的id，而不见得是实际id。

您好，这次仔细检查aarch64发现在SMP模式，非0核为主核启动下，逻辑上有问题，无法正确适配。

假设cpu核心的启动顺序为2，0，1，3。gic地址以0核为基地址，1，2，3核依次增加偏移值

[image]
启动过程中，2核首先进入以上函数，cpu_id= rt_hw_cpu_id；按照现在的适配方法，得到的结果为cpu_id= 2。redist_base参数传入0核基地址，这样就正确适配了2核和0核，其他核心进入这个函数后，理论上也进行了正确的适配。
[image]
但是如上图所示，在这里 如果时2核启动时进入 rt_hw_cpu_id（）因为结果等于2，就进行了错误的处理。这里把0定死为主核，这样矛盾的地方还有很多

BernardXiong, 2024-03-11:
唔，目前在推进的方式对于非SMP模式，以AMP模式来启动的，基本上都没考虑到呢。

closed 2024-04-11 by messigogogo