RT-Thread/rt-thread #8592: 需要整理的一些细节点

mysterywolf, 2024-03-09:
- 部分BSP的HAL库需要拆成软件包的方式，不能把HAL库嗷嗷往主仓提交，例如STM32的HAL库
- pin框架read返回值从rt_uint8_t 改为rt_ssize_t
- 使用os.join.path函数，而不是直接用/连接路径

BernardXiong, 2024-03-11:
发布env v2.0，启用python中的venv环境，并提供软件包下载功能。

wirano, 2024-04-08:
不能再同意第二条了😂

pineapple-man, 2024-04-12:
stm32中的HAL库真的太多了,感觉BSP就显的很乱

Rbb666, 2024-04-14:
> stm32中的HAL库真的太多了,感觉BSP就显的很乱

库文件后面应该都放到软件包里面，减少体积

chnykn, 2024-04-15:
第二条非常有必要

meng-plus, 2024-04-23:
整个仓库都大的太恐怖了全是bsp占用

charlown, 2024-04-29:
怎么拆分厂家的驱动库做成软件包的形式，有个范例就好了
ch32 ci看护是arm还是risc-v出了问题么？沁恒最近在risc-v方面大力发展, 他们arm的芯片很久没出新的

mr-cn, 2024-06-05:
建议BSP把具体的board也拆分出去，主仓库保留对chip的支持就好了

dujunqiu, 2024-06-26:
stm32用的HAL库，gd32用的标准库，能统一一下就好了

meng-plus, 2024-07-05:
我最近在玩repo 我认为他是个很好的多仓库管理工具，是否考虑往这个方向发展下 
即使同一个芯片的HAL库官方也在持续更新，拆包有助于精简仓库占用，独立更好的维护每个组件

GPTKEY, 2024-07-08:
可以考虑：对于STM32 直接修改构建脚本， 去掉BSP中的HAL库，直接使用CubeMX生成的HAL库，这样兼容性更强

dujunqiu, 2024-07-08:
> 我最近在玩repo 我认为他是个很好的多仓库管理工具，是否考虑往这个方向发展下 即使同一个芯片的HAL库官方也在持续更新，拆包有助于精简仓库占用，独立更好的维护每个组件

好主意，我看NXP的github库，已经拆分成多个子模块独立管理了

wdfk-prog, 2024-07-09:
> 可以考虑：对于STM32 直接修改构建脚本， 去掉BSP中的HAL库，直接使用CubeMX生成的HAL库，这样兼容性更强

- 的确如此;甚至对于驱动中有关初始化配置的部分,都可以去除;保留一些常用配置修改既可;
- 不然对于每个系列都要兼容,不仅代码难看,而且维护起来也不容易,加个功能容易在其他没验证的系列出问题
- 但是如果是CubeMx生成的Hal库,就需要用户对驱动有意识,自行进行正确的配置

sulfurandcu, 2024-10-23:
- rt-thread repo
- bsp repo
  - stm32
  - gd32
  - ...
  - rt-thread @ master

example: https://github.com/sisong/HPatchLite/tree/main

↑这么搞有什么问题/难度吗？#9519

mysterywolf, 2024-10-23:
submodule方式无法进行gitee同步备份，国内拉取会出问题的。

aozima, 2024-10-24:
> submodule

仅技术上回复下：
git submodule的url可以使用相对地址，这样不管源地址是ssh还是https，或是换了服务器域，只要相对路径没变就能引用到。

yuqingli05, 2024-12-20:
> 建议BSP把具体的board也拆分出去，主仓库保留对chip的支持就好了

老早前我提过，被否了。现在越来越大，很多用户只是做自己的工程，我感觉更有必要了。主仓库留几个经典案例就行了

ThinkCodeStudio, 2025-01-03:
建议把BSP拆分出去, 然后使用时按需下载, 这么多芯片也用不过来; 除去我要使用的芯片, 其他芯片BSP已经占1G多了

Elsa-is-My-Muse, 2025-02-20:
建议env支持使用scons工具生成compile_command.json文件

StrandingNeko, 2025-02-20:
> 建议env支持使用scons工具生成compile_command.json文件

scons --cdb
scons --target=vsc