RT-Thread/rt-thread #8775: 使用RT-Studio生产的STM32F407VE工程运行一段时间后自动停止

CN-CaoJun, 2024-04-11:
在RT-studio中根据默认的设置新建如下工程时，可以启动板子，但是运行300s左右后就会停止，即使配置使用外部时钟，依旧如此。
工具及配置过程：
RT-studio版本 2.2.7
BSP 5.0.2
Hardware STM32F407VET6 
[image]

运行过程：
[image]

但是每次运行接近300s之后就会卡住，确认不是串口工具的原因，需要重新插拔电源才行。
请问下这个是什么情况，我有什么没有设置吗？
[image]

chnykn, 2024-04-12:
[image]
正好 我这里有一个STM32F407VE，试了一下，并不会出错。

Rbb666, 2024-04-14:
可以debug看看，在卡死的时候停止调试看看停在哪里，再看看为什么不切换到目标线程

CN-CaoJun, 2024-04-14:
感谢各位，有时间了再试下我再来更新

CN-CaoJun, 2024-04-14:
这个问题在我这边是必现的，当前出现的情况是：306个tick之后就没有打印了，在调试界面暂停的时候，发现任务就在IDLE task里面。
[image]
[image]

CN-CaoJun, 2024-04-14:
> 正好 我这里有一个STM32F407VE，工具及配置与你的完全一样。试了一下，并不会出错。 没有任何特殊的设置，除了我用的ST-Link；貌似 你用的DAP-Link。理论上，应该不是这个原因。
> 
> ![1](https://private-user-images.githubusercontent.com/4152089/321918167-4b5f969d-6549-4fe6-a8f9-b9d3f6957ba9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxMDI1MTIsIm5iZiI6MTcxMzEwMjIxMiwicGF0aCI6Ii80MTUyMDg5LzMyMTkxODE2Ny00YjVmOTY5ZC02NTQ5LTRmZTYtYThmOS1iOWQzZjY5NTdiYTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTRUMTM0MzMyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2Y0ZTA0YmM0MzE3ZGJkNWNiZjYyZjczNzRmYjA0YmIzYmVkNGZjMTZlZTYyYTI1YTVjZTg2ZTQ5M2U1NWQxZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.3FLU6Psxs8qM1Iq2EqluCIxP5WzuAjG3UtjHtLBNWj0)

麻烦您这边多跑一段时间看看，thx。

CN-CaoJun, 2024-04-14:
> 可以debug看看，在卡死的时候停止调试看看停在哪里，再看看为什么不切换到目标线程

还在学习RTTHREAD中，希望我能弄清楚一下为啥切不了目标线程

chnykn, 2024-04-15:
> 麻烦您这边多跑一段时间看看，thx。

图我就不贴了，刚刚连续跑了一个多小时，不会出错！

CN-CaoJun, 2024-04-15:
Update:
今天工作之余再调试了下，
在debug模式下跑：
跑一段时间之后，调试器自己会在idle里面停一下， 然后run可以继续跑下去，但是需要串口监控软件重置下才能打印出接下来的log。
[image]

在正常模式下跑：
跑一段时间后串口没有输出，重置串口后可以看到接下来的打印。使用的串口软件是ScureCRT

因此我认为可以把这个问题的描述调整成：
使用RT-Studio生产的STM32F407VE工程运行一段时间后串口无法输出，但是重置串口软件后可以看到后续的输出。

Rbb666, 2024-04-15:
> Update: 今天工作之余再调试了下， 在debug模式下跑： 跑一段时间之后，调试器自己会在idle里面停一下， 然后run可以继续跑下去，但是需要串口监控软件重置下才能打印出接下来的log。 ![image](https://private-user-images.githubusercontent.com/127820339/322412031-eec104b7-6119-479f-92a8-ef3a78c6ed85.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxNzA4OTgsIm5iZiI6MTcxMzE3MDU5OCwicGF0aCI6Ii8xMjc4MjAzMzkvMzIyNDEyMDMxLWVlYzEwNGI3LTYxMTktNDc5Zi05MmE4LWVmM2E3OGM2ZWQ4NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxNVQwODQzMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03OThiMWU2MjY1NzFhZDQxNjE0ODZmMjFmZWVhOWJlZmZjNzA3ZGNjYzBhYmY4YWZjM2MwODAxZDI1YTRiNzZmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.M4kTbPtbb8bi5zS2KqKCrbwAJvQlZK1iNWVJ8Txf9PU)
> 
> 在正常模式下跑： 跑一段时间后串口没有输出，重置串口后可以看到接下来的打印。使用的串口软件是ScureCRT
> 
> 因此我认为可以把这个问题的描述调整成： 使用RT-Studio生产的STM32F407VE工程运行一段时间后串口无法输出，但是重置串口软件后可以看到后续的输出。
> 
> 我认为这对我也是值得研究的，因为同样的环境在监控其他软件的串口输出时并没有出现类似的情况，接下来看下是什么打断了串口的输出。

换个串口助手试试呢？比如我使用的是mobaxterm

CN-CaoJun, 2024-04-15:
> 换个串口助手试试呢？比如我使用的是mobaxterm

Mobaxterm也是同样的情况，这个现象非常奇妙，😂

CN-CaoJun, 2024-04-17:
换了开发的电脑和串口之后，发现问题不再复现😂，感谢支持！

closed 2024-04-17 by CN-CaoJun