RT-Thread/rt-thread #8785: rt-thread/bsp/qemu-vexpress-a9下编译失败

pineapple-man, 2024-04-12:
我在https://github.com/RT-Thread/rt-thread/tree/d5aff0c963057abd5ebee3557ddf84f1a4e3e0fd这个版本下,在`rt-thread/bsp/qemu-vexpress-a9`下使用`scons`命令进行编译时出现如下问题
```
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/applications/lcd_init.o
CC build/applications/main.o
CC build/applications/mnt.o
CC build/drivers/board.o
CC build/drivers/drv_keyboard.o
CC build/drivers/drv_mouse.o
CC build/drivers/drv_sdio.o
CC build/drivers/drv_timer.o
CC build/drivers/secondary_cpu.o
CC build/drivers/serial.o
CC build/kernel/components/dfs/dfs_v2/filesystems/devfs/devfs.o
/home/mingchaohu/rt-thread/components/dfs/dfs_v2/filesystems/devfs/devfs.c: In function 'dfs_devfs_open':
/home/mingchaohu/rt-thread/components/dfs/dfs_v2/filesystems/devfs/devfs.c:34:32: error: 'O_DIRECTORY' undeclared (first use in this function); did you mean 'FT_DIRECTORY'?
   34 |             && !(file->flags & O_DIRECTORY))
      |                                ^~~~~~~~~~~
      |                                FT_DIRECTORY
/home/mingchaohu/rt-thread/components/dfs/dfs_v2/filesystems/devfs/devfs.c:34:32: note: each undeclared identifier is reported only once for each function it appears in
scons: *** [build/kernel/components/dfs/dfs_v2/filesystems/devfs/devfs.o] Error 1
scons: building terminated because of errors.
```
似乎存在部分的配置错误?

pineapple-man, 2024-04-12:
貌似在另一台主机并没有问题

closed 2024-04-13 by pineapple-man

pineapple-man, 2024-06-26:
> 貌似在另一台主机并没有问题

目前找到的原因是因为 arm 编译器的问题，在最新的arm编译器中（12.*, 13.*）都会导致RTT编译失败，使用11.2就可以正常编译

ThearchyHelios, 2025-07-22:
> > 貌似在另一台主机并没有问题
> 
> 目前找到的原因是因为 arm 编译器的问题，在最新的arm编译器中（12._, 13._）都会导致RTT编译失败，使用11.2就可以正常编译

我这边还是出现了同样的问题，
```$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14)) 11.2.1 20220111
```

---

```
$ scons             
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/applications/lcd_init.o
CC build/applications/main.o
CC build/applications/mnt.o
CC build/drivers/board.o
CC build/drivers/drv_keyboard.o
CC build/drivers/drv_mouse.o
CC build/drivers/drv_sdio.o
CC build/drivers/drv_timer.o
CC build/drivers/drv_uart.o
CC build/drivers/secondary_cpu.o
CC build/kernel/components/dfs/dfs_v2/filesystems/devfs/devfs.o
/Users/???/Documents/rt-thread/components/dfs/dfs_v2/filesystems/devfs/devfs.c: In function 'dfs_devfs_open':
/Users/???/Documents/rt-thread/components/dfs/dfs_v2/filesystems/devfs/devfs.c:34:32: error: 'O_DIRECTORY' undeclared (first use in this function); did you mean 'FT_DIRECTORY'?
   34 |             && !(file->flags & O_DIRECTORY))
      |                                ^~~~~~~~~~~
      |                                FT_DIRECTORY
/Users/???/Documents/rt-thread/components/dfs/dfs_v2/filesystems/devfs/devfs.c:34:32: note: each undeclared identifier is reported only once for each function it appears in
scons: *** [build/kernel/components/dfs/dfs_v2/filesystems/devfs/devfs.o] Error 1
scons: building terminated because of errors.
```

```
$ sw_vers
ProductName:		macOS
ProductVersion:		15.5
BuildVersion:		24F74
```