RT-Thread/rt-thread #8786: bsp/renesas/rzt2m_rsk bsp目录是否可以优化一下？

pineapple-man, 2024-04-12:
目前看到这个rzt2m_rsk这个目录下的内容非常的奇怪，本身就是一个bsp的目录，为什么还会存在`bsp/renesas/rzt2m_rsk/rzt/fsp/src/bsp`这样的路径，在bsp中嵌套bsp？这是什么操作？

pineapple-man, 2024-04-12:
而且我怎么感觉这个bsp下的内容其实并没有使用libcpu/arm/cortex-r52下面的相关内容，反而使用的是`bsp/renesas/rzt2m_rsk/rzt/fsp/src/bsp/cmsis/Device/RENESAS/Source/cr/startup_core.c`作为整个bsp中cpu中的相关初始化行为？

pineapple-man, 2024-04-12:
我将编译出来的内容，反编译出来，在反编译的汇编文件中并没有看到libcpu中的内容（vector_gcc.S以及start_gcc.S），所以在链接的过程中，是不是将`libcpu/arm/cortex-r52`的中断表，startup code完全抛弃了，然后使用`bsp/renesas/rzt2m_rsk/rzt/fsp/src/bsp/cmsis/Device/RENESAS/Source/cr`中的startup code

kurisaW, 2024-04-14:
> 目前看到这个rzt2m_rsk这个目录下的内容非常的奇怪，本身就是一个bsp的目录，为什么还会存在`bsp/renesas/rzt2m_rsk/rzt/fsp/src/bsp`这样的路径，在bsp中嵌套bsp？这是什么操作？

这个主要是瑞萨fsp生成的原厂驱动，只是使用同一个名字，我们不能说RT-Thread的这个bsp目录是班级支持包，那其他厂商在设置文件层次的时候就得避免这种情况，他们的这个目录`bsp/renesas/rzt2m_rsk/rzt/fsp/src/bsp`只是裸机的一些配置信息和CMSIS标准，两者不是同一个概念和用途

kurisaW, 2024-04-14:
> 而且我怎么感觉这个bsp下的内容其实并没有使用libcpu/arm/cortex-r52下面的相关内容，反而使用的是`bsp/renesas/rzt2m_rsk/rzt/fsp/src/bsp/cmsis/Device/RENESAS/Source/cr/startup_core.c`作为整个bsp中cpu中的相关初始化行为？

正如你所言，由于瑞萨原厂那边已经提供了startup_core.c来实现中断向量表和异常处理，我们的libcpu事实上是去接住这个向量表，然后对异常做了一层处理，用于打印异常发生的堆栈信息和栈回溯支持，这个你应该重新梳理一遍工程源码

pineapple-man, 2024-04-15:
> > 而且我怎么感觉这个bsp下的内容其实并没有使用libcpu/arm/cortex-r52下面的相关内容，反而使用的是`bsp/renesas/rzt2m_rsk/rzt/fsp/src/bsp/cmsis/Device/RENESAS/Source/cr/startup_core.c`作为整个bsp中cpu中的相关初始化行为？
> 
> 正如你所言，由于瑞萨原厂那边已经提供了startup_core.c来实现中断向量表和异常处理，我们的libcpu事实上是去接住这个向量表，然后对异常做了一层处理，用于打印异常发生的堆栈信息和栈回溯支持，这个你可以重新梳理一遍工程源码

OK，我大致上理解了这种设计思路

closed 2024-04-15 by pineapple-man