RT-Thread/rt-thread #8863: [Kconfig] 是不是放弃“option env=”了？

sp-cai, 2024-04-26:
发现自 #8664 之后，所有 BSP 里 Kconfig 的 “option env=” 特性都无效了，
进一步对比以前的代码，是 kconfiglib.py 的函数“_parse_block”处少了如下代码：
                if node.item.env_var:
                    if node.item.env_var in os.environ:
                        os.environ[node.item.name] = os.environ[node.item.env_var]
                    else:
                        os.environ[node.item.name] = ((node.defaults[0])[0]).name

此特性还要用吗？是不是以后还会修复？如果已经永久停用了，是不是应该清理所有 BSP 里有关内容？

BernardXiong, 2024-04-26:
需要首先验证的是，如果放弃`option env=`的方式，studio是否没问题。如果没问题，那么就可以考虑放弃了。当然也有一种方式是修改kconfiglib的代码，但这个是否会涉及到上游代码了。

sp-cai, 2024-04-27:
> 需要首先验证的是，如果放弃`option env=`的方式，studio是否没问题。如果没问题，那么就可以考虑放弃了。当然也有一种方式是修改kconfiglib的代码，但这个是否会涉及到上游代码了。

上游有他的考量，想在上游加上恐怕指望不上了。如果还有用，也可以考虑在其它地方扩展。
以下是来自上游(Kconfiglib)可能相关的信息以供参考：
`
Environment variables are expanded directly in e.g. and statements, meaning symbols are redundant.sourcemainmenuoption env

This is the standard behavior with the new [Kconfig preprocessor](https://github.com/torvalds/linux/blob/master/Documentation/kbuild/kconfig-macro-language.rst), which Kconfiglib implements.

option env symbols are accepted but ignored, which leads the caveat that they must have the same name as the environment variables they reference (Kconfiglib warns if the names differ). This keeps Kconfiglib compatible with older Linux kernels, where the name of the symbol always matched the environment variable. Compatibility with older Linux kernels is the main reason is still supported.option envoption env

The C tools have dropped support for .option env
`

BernardXiong, 2024-04-27:
> > 需要首先验证的是，如果放弃`option env=`的方式，studio是否没问题。如果没问题，那么就可以考虑放弃了。当然也有一种方式是修改kconfiglib的代码，但这个是否会涉及到上游代码了。
> 
> 上游有他的考量，想在上游加上恐怕很难指望了。如果还有用，也可以考虑在其它地方扩展。 以下是来自上游[`(Kconfiglib)`](https://github.com/ulfalizer/Kconfiglib)可能与之相关的信息以供参考：
> 
> * Environment variables are expanded directly in e.g. `source` and `mainmenu` statements, meaning `option env` symbols are redundant.
>   This is the standard behavior with the new [`Kconfig preprocessor`](https://github.com/torvalds/linux/blob/master/Documentation/kbuild/kconfig-macro-language.rst),which Kconfiglib implements.
>   `option env` symbols are accepted but ignored, which leads the caveat that they must have the same name as the environment variables they reference (Kconfiglib warns if the names differ). This keeps Kconfiglib compatible with older Linux kernels, where the name of the `option env` symbol always matched the environment variable. Compatibility with older Linux kernels is the main reason `option env` is still supported.
>   The C tools have dropped support for `option env`.
> 
> 文中意思似乎是已经放弃支持`option env`了，转而推新的 Kconfig 预处理标准，如：
> 
> ```
> BSP_DIR := ../../..
> source "$(RTT_DIR)/Kconfig"
> ```

那就需要验证studio中的web kconfig是否支持这样的方式了。

sp-cai, 2024-06-21:
#9050

closed 2024-06-21 by sp-cai