RT-Thread/rt-thread #8919: 在bsp/qemu-virt64-aarch64示例中使能umqtt后无法subscribe

kaypton, 2024-05-08:
rt-thread内核使能了smart，网络使能LwIP，均测试没问题。在online packages里边使能umqtt，然后选择编译示例程序（umqtt_ex_start等）。最后使用未经修改的qemu.sh脚本启动rt-thread。

在msh中键入umqtt_ex_start后rtt成功连接到后端mosquitto（mosquitto经测试配置没问题，freertos的coremqtt可以正常交互），但是当键入umqtt_ex_subscribe topic/test后输出：
[E/umqtt] subscribe recv message timeout! topic: topic/test
从而导致无法正常订阅MQTT主题。

BernardXiong, 2024-05-08:
请在用户态来使用

kaypton, 2024-05-08:
> 请在用户态来使用

其实我需要在内核态使用一个轻量的MQTT，用户态有其他用处。您的意思是umqtt库在启用smart的时候只能在userapp中使用么？
刚刚尝试关闭了smart，然后使用最新的rtt 5.1.0 bsp/qemu-virt64-aarch64进行上述测试仍然遇到相同的问题。

BernardXiong, 2024-05-10:
> > 请在用户态来使用
> 
> 其实我需要在内核态使用一个轻量的MQTT，用户态有其他用处。您的意思是umqtt库在启用smart的时候只能在userapp中使用么？ 刚刚尝试关闭了smart，然后使用最新的rtt 5.1.0 bsp/qemu-virt64-aarch64进行上述测试仍然遇到相同的问题。

注意下工具，是否是libc引起的问题