RT-Thread/rt-thread #8930: duo : gpio : rt_pin_read 无法读出引脚的状态

unicornx, 2024-05-10:
`bsp/cvitek/cv18xx_risc-v/` 最新 master
----

在编写跑马灯例子过程中，尝试读取 GPIO 引脚的状态值，失败

简单示例代码如下：
```c
static int led_on(void)
{
    rt_pin_write(pinno, PIN_HIGH);
    rt_kprintf("%d\n", rt_pin_read(pinno)); // 这里希望读到 1，但是 0
    return 0;
}

static int led_off(void)
{
    rt_pin_write(pinno, PIN_LOW);
    rt_kprintf("%d\n", rt_pin_read(pinno)); // 这里希望读到 0，
    return 0;
}
```

看了一下驱动，rt_pin_read 回去读取 GPIO_EXT_PORTA，怀疑这个寄存器是不是反应了引脚的状态？

unicornx, 2024-05-10:
已经 report 给 sophgo 看看这个寄存器到底怎么个说法

BernardXiong, 2024-05-11:
是否可能和pin的模式相关？难道需要先配置成输入模式？

flyingcys, 2025-01-05:
#define INPUT_PIN     "A14"
rt_uint16_t input = rt_pin_get(INPUT_PIN);
rt_pin_mode(input, PIN_MODE_INPUT_PULLUP);
rt_kprintf("Input pin value: %d\n", rt_pin_read(input));
读取pin输入状态正常

flyingcys, 2025-01-05:
#define INPUT_PIN     "A14"
    rt_uint16_t input = rt_pin_get(INPUT_PIN);
    rt_pin_mode(input, PIN_MODE_INPUT_PULLUP);
        rt_kprintf("Input pin value: %d\n", rt_pin_read(input));
读取gpio输入正常。

unicornx, 2025-01-07:
这个 issue 先关了吧，时间太长了，以后再遇到再看

closed 2025-01-07 by unicornx