RT-Thread/rt-thread #8939: duo：编译时是否可以默认加入

unicornx, 2024-05-11:
有一个疑问是我发现在 `bsp/cvitek/drivers/SConscript` 中 
```
if GetDepend('BSP_USING_CV18XX'):
    src += ['drv_gpio.c']
```

所以我想先确认一下对于 GPIO 这个，原先的设计是怎么考虑的，我们是否可以直接将其和 UART 一样对待，即默认加入 `drv_gpio.c`，参考 `bsp/allwinner/libraries/drivers/SConscript`, 默认加入了  'drv_uart.c' 和 'drv_pin.c'

BernardXiong, 2024-05-12:
可以默认加入的，可以是打开配置（menuconfig），然后把.config提交上来即可默认加入。

unicornx, 2024-05-13:
> 可以默认加入的，可以是打开配置（menuconfig），然后把.config提交上来即可默认加入。

这个 issue 希望讨论的是 对于 duo，gpio 驱动是否可以作为默认支持的驱动加在配置里。不是讨论具体的加入的技术问题，是需求和设计问题。

BernardXiong, 2024-05-13:
> > 可以默认加入的，可以是打开配置（menuconfig），然后把.config提交上来即可默认加入。
> 
> 这个 issue 希望讨论的是 对于 duo，gpio 驱动是否可以作为默认支持的驱动加在配置里。不是讨论具体的加入的技术问题，是需求和设计问题。

可以加入。需求其实是使用者希望默认获得什么，所以实际情况是使用这个bsp的开发者希望如何。目前应该使用的不多，所以你如果有需求，则可以默认开启加入。

flyingcys, 2024-05-14:
https://github.com/RT-Thread/rt-thread/pull/8946已修复该问题

closed 2024-05-14 by unicornx