RT-Thread/rt-thread #8943: duo: atoi crash

unicornx, 2024-05-13:
简单的调用 atoi 函数会导致异常：

```c
#include <rtthread.h>
#include <stdio.h>
#include <stdlib.h>

static int test(int argc, char *argv[])
{
    int v;
    v = atoi("100");
    rt_kprintf("v is %d\n", v);

    return RT_EOK;
}
MSH_CMD_EXPORT(test, test);
```

log 如下：
```
Unhandled Exception 2:Illegal Instruction
scause:0x0x0000000000000002,stval:0x0x0000000000000000,sepc:0x0x00000000802001ce
--------------Dump Registers-----------------
Function Registers:
        ra(x1) = 0x0x0000000080218002   user_sp = 0x0x0000000080259040
        gp(x3) = 0x0x0000000080234bb8   tp(x4) = 0x0x00000000deadbeef
Temporary Registers:
        t0(x5) = 0x0x0000000000000120   t1(x6) = 0x0x00000000deadbeef
        t2(x7) = 0x0x0000000080258f30
        t3(x28) = 0x0x00000000deadbeef  t4(x29) = 0x0x00000000deadbeef
        t5(x30) = 0x0x00000000deadbeef  t6(x31) = 0x0x00000000deadbeef
Saved Registers:
        s0/fp(x8) = 0x0x0000000080259080        s1(x9) = 0x0x00000000deadbeef
        s2(x18) = 0x0x00000000deadbeef  s3(x19) = 0x0x00000000deadbeef
        s4(x20) = 0x0x00000000deadbeef  s5(x21) = 0x0x00000000deadbeef
        s6(x22) = 0x0x00000000deadbeef  s7(x23) = 0x0x00000000deadbeef
        s8(x24) = 0x0x00000000deadbeef  s9(x25) = 0x0x00000000deadbeef
        s10(x26) = 0x0x00000000deadbeef s11(x27) = 0x0x00000000deadbeef
Function Arguments Registers:
        a0(x10) = 0x0x000000008022c7f8  a1(x11) = 0x0x00000000802590a8
        a2(x12) = 0x0x00000000802590a8  a3(x13) = 0x0x00000000802590a8
        a4(x14) = 0x0x0000000000000009  a5(x15) = 0x0x0000000000000009
        a6(x16) = 0x0x0000000000000074  a7(x17) = 0x0x00000000deadbeef
sstatus = 0x0x0000000200040120
        Supervisor Interrupt Disabled
        Last Time Supervisor Interrupt Enabled
        Last Privilege is Supervisor Mode
        Permit to Access User Page
        Not Permit to Read Executable-only Page
satp = 0x0x0000000000000000
        Current Page Table(Physical) = 0x0x0000000000000000
        Current ASID = 0x0x0000000000000000
        Mode = No Address Translation/Protection Mode
-----------------Dump OK---------------------
--------------Thread list--------------
current thread: tshell
--------------Backtrace--------------

```

flyingcys, 2024-05-20:
v5.1.0正常，master有问题

unicornx, 2024-06-03:
网上看到一篇文章，可能和这个问题有关系，just FYI：https://zhuanlan.zhihu.com/p/619893693

unicornx, 2024-06-09:
当前 小核和 大核的 gcc 用的都是 Xuantie-900-gcc-elf-newlib-x86_64-V2.8.1，但是大核需要使用 musl 的gcc，对大核换成 musleabi 的之后就不会 crash 了。

后面具体怎么解，还要看看能否将大核和小核的工程尽量统一起来。参考 https://github.com/RT-Thread/rt-thread/pull/9028 的讨论。

unicornx, 2024-06-24:
> v5.1.0正常，master有问题

5.1.0 应该也不行吧

unicornx, 2024-06-24:
小核目前用 newlib gcc 是好的。

unicornx, 2024-07-18:
打算将 cv18xx_riscv-v 默认切换到 musl gcc。即修改 bsp/cvitek/cv18xx_risc-v/rtconfig.py 中对 EXEC_PATH 和 PREFIX， 模仿 bsp/qemu-virt64-riscv/rtconfig.py

unicornx, 2024-07-18:
> RFC:
> 
> 打算将 cv18xx_riscv-v 默认切换到 musl gcc。即修改 bsp/cvitek/cv18xx_risc-v/rtconfig.py 中对 EXEC_PATH 和 PREFIX， 模仿 bsp/qemu-virt64-riscv/rtconfig.py

polarvid, 2024-07-18:
libc 的差别我理解应该没有太严重的影响，特别是系统内核这边用到的主要是字符处理、数学库这类的。

unicornx, 2024-07-19:
一个有趣的发现：

目前主线上自从 ”beee77f372 feat: support ARCH_REMAP_KERNEL on libcpu/c906 (#9123)" 合入后，atoi 的问题就不见了。使用 newlib gcc 一样工作正常。

unicornx, 2024-07-19:
关闭这个 issue，目前 master 已经不会出错。

closed 2024-07-19 by unicornx