RT-Thread/rt-thread #8948: ..\..\..\components\drivers\ipc\completion.c文件缺失

nesa1978, 2024-05-14:
在编译BSP中示例时报错"no source": Error:  #5: cannot open source input file "..\..\..\components\drivers\ipc\completion.c": No such file or directory，找到对应的路径下没有对应的文件
[image]
[image]

Yumbrella, 2024-05-15:
佬解决了吗

nesa1978, 2024-05-15:
> 佬解决了吗
下载旧版本的Rtthread就行了5.0.1和5.0.2都可以

BernardXiong, 2024-05-15:
改名调整了，需要重写刷新下工程文件吧

guanjianhe, 2024-05-15:
1. 在工程删除 completion.c 文件
2. 添加 completion_comm.c completion_up.c condvar.c 3个文件
3. 添加cpu_up.c 文件 @nesa1978 @Yumbrella

mysterywolf, 2024-05-15:
重新用env 用命令 scons --target=mdk5即可

closed 2024-05-25 by nesa1978

dujunqiu, 2024-06-20:
我还奇怪官方例程怎么缺少.c文件

shandianchengzi, 2024-06-30:
> scons --target=mdk5

 scons --target=mdk5解决的

dujunqiu, 2024-06-30:
这里有修改说明
https://blog.csdn.net/dujunqiu/article/details/139826672?spm=1001.2014.3001.5501

Fcheng-1, 2024-08-15:
STM32L4系列的library库里面，都没有L4的HAL库，还得手动添加，建议代码，keil工程提交，先编译一下能否编过。