RT-Thread/rt-thread #8969: duo: cv18xx_risc-v failed to build

unicornx, 2024-05-21:
直接 clone 下来的版本，进入  `bsp/cvitek/cv18xx_risc-v/` 后运行 scons
build log output:
```
board/board.c: In function 'primary_cpu_entry':
board/board.c:58:5: warning: implicit declaration of function 'rt_assert_set_hook'; did you mean 'rt_free_sethook'? [-Wimplicit-function-declaration]
   58 |     rt_assert_set_hook(__rt_assert_handler);
      |     ^~~~~~~~~~~~~~~~~~
      |     rt_free_sethook
......
LINK rtthread.elf
/opt/Xuantie-900-gcc-elf-newlib-x86_64-V2.8.1/bin/../lib/gcc/riscv64-unknown-elf/10.4.0/../../../../riscv64-unknown-elf/bin/ld: build/board/board.o: in function `primary_cpu_entry':
/home/u/ws/duo/rt-thread/bsp/cvitek/cv18xx_risc-v/board/board.c:58: undefined reference to `rt_assert_set_hook'
Memory region         Used Size  Region Size  %age Used
            SRAM:      548808 B        32 MB      1.64%
collect2: error: ld returned 1 exit status
scons: *** [rtthread.elf] Error 1
scons: building terminated because of errors.
```

但是如果运行过 `scons --menuconifg` 保存后再编译就 ok

初步原因分析，默认的 .config 文件中缺少 `RT_DEBUGING_ASSERT` 的定义导致告警和链接失败。

BernardXiong, 2024-05-21:
是的，这类情况下需要刷新下.config文件

BernardXiong, 2024-05-21:
@polarvid 未来可以多考虑下，如果这个宏没有会是什么样的，是否可能出现错误。所以有的时候，可以做个宏检查，如果没有则进行定义。通过系列的方式，可以做到兼容

unicornx, 2024-06-13:
暂时先这么操作吧，解决方法就是每次 clone 下来的仓库，先执行一遍 `scons --menuconfig` 并 save 后再 scons。

关闭此 issue。

closed 2024-06-13 by unicornx

polarvid, 2024-06-13:
可以更新一下 .config 与 rtconfig.h 文件，推到仓库上去。