RT-Thread/rt-thread #8978: duo: little core does not work

unicornx, 2024-05-22:
最新主线上的 duo 的小核无法工作，回退到 5.1.0 正常

unicornx, 2024-05-23:
`[bsp/cvitek]update gpio driver (#8946)` 这笔 patch 会导致小核 crash，gpio 部分的代码似乎有问题。

adrain-yu, 2024-05-24:
RT_HW_HEAP_BEGIN:83f74300 RT_HW_HEAP_END:84000000 size: 572672

 \ | /
- RT -     Thread Operating System
 / | \     5.2.0 build May 24 2024 14:01:36
 2006 - 2024 Copyright by RT-Thread team
 小核 只能打印这些 shell和Hello 打印不出来是为啥嘞

unicornx, 2024-07-15:
貌似主线上的这个问题并没有解决，看到的现象和 @adrain-yu 的一样。

unicornx, 2024-07-15:
这个问题和 #8889 有关

该 pr 在 `libcpu/risc-v/common/trap_common.c` 中定义了一个 weak 的函数 handle_trap, 这个函数和 libcpu/risc-v/rv64/trap.c 中的  handle_trap 同名，这导致对于 `CPU         ='rv64'` 的产品的 handle_trap 会被同名的函数覆盖掉。

unicornx, 2024-07-15:
[image]

flyingcys, 2024-07-15:
如果bsp里面实现了trap_entry，libcpu里面也有，应该报错重复定义才对，编译不过。
是不是把trap_entry加上weak就可以了

unicornx, 2024-07-15:
> 如果bsp里面实现了trap_entry，libcpu里面也有，应该报错重复定义才对，编译不过。 是不是把trap_entry加上weak就可以了

#8889 那笔修改对 trap_entry 是加了 weak 的，trap_entry 上加了 weak 后就不会有问题，问题是出在 handle_trap 上，他加的那个 common 里的 handle_trap 和 rv64 里那个 同名，而且都是 weak，所以对 duo，gcc 编译时选择了 common 里的，导致了问题

unicornx, 2024-07-16:
fixed in #9189

closed 2024-07-16 by unicornx