RT-Thread/rt-thread #8982: [bsp/rk3568]:使用v5.1.0版本无法正常启动

Theshy25, 2024-05-24:
使用v5.0.2版本可以正常启动，确认串口波特率正常，切换至v5.1.0版本无法正常启动，https://github.com/RT-Thread/rt-thread/commit/249871cbbcdc14f1c16a2d502639780ebc6ae078#diff-70d96506e5aae17bbe4bdeb030ad8a4902ed541a4f5e1751030bf4e3f95b8a03合入了该commit后无法正常启动。
[image]

Theshy25, 2024-05-24:
一旦合入上述的commit即无法正常启动

Theshy25, 2024-05-28:
该平台使用独立的板卡初始化函数rt_hw_board_init（rt-thread/bsp/rockchip/rk3568/driver/board.c），在调用rt_hw_mmu_map_init初始化rt_mmu_info结构体时，使用的虚拟内存地址区间不满足初始化的内存区间判断，导致rt_page_init初始化失败运行至ASSERT处

Theshy25, 2024-05-28:
rt_hw_mmu_map_init(&rt_kernel_space, (void *)0xffffd0000000, 0x10000000, MMUTable, 0);使用rt-thread/libcpu/aarch64/common/setup.c下初始化rt_mmu_info结构体的虚拟地址空间的虚拟地址的起始地址与大小，即可正常启动

BernardXiong, 2024-05-30:
> ```
> rt_hw_mmu_map_init(&rt_kernel_space, (void *)0xffffd0000000, 0x10000000, MMUTable, 0);使用rt-thread/libcpu/aarch64/common/setup.c
> ```
> 下初始化rt_mmu_info结构体的虚拟地址空间的虚拟地址的起始地址与大小，即可正常启动

欢迎PR修正

dmanzhang, 2024-11-12:
使用上述代码后可以启动

Starting kernel ...

heap: [0x2038a000 - 0x2438a000]

 \ | /
- RT -     Thread Operating System
 / | \     5.1.0 build Nov 12 2024 10:09:29
 2006 - 2024 Copyright by RT-Thread team
lwIP-2.1.2 initialized!
[D/drv.emac] emac open
[D/drv.emac] emac device init success
[I/sal.skt] Socket Abstraction Layer initialize success.
Hi, this is RT-Thread!!
msh />[D/drv.emac] link up
[D/drv.emac] speed: 1000M
ifconfig e0 192.168.11.55 192.168.11.1 255.255.255.0
config : e0
IP addr: 192.168.11.55
Gateway: 192.168.11.1
netmask: 255.255.255.0
msh />ps
thread   cpu bind pri  status      sp     stack size max used left tick   error  tcb addr
-------- --- ---- ---  ------- ---------- ----------  ------  ---------- ------- ------------------
tshell     0   4   20  running 0x00000538 0x00001000    51%   0x0000000a OK      0x0000000020393230
phy      N/A   4   12  suspend 0x00000418 0x00001000    33%   0x00000002 EINTRPT 0x0000000020391980
sys wor  N/A   4   23  suspend 0x000003b0 0x00000800    54%   0x0000000a OK      0x0000000020390e78
tcpip    N/A   4   10  suspend 0x00000490 0x00004000    11%   0x00000014 EINTRPT 0x000000002038cc98
etx      N/A   4   12  suspend 0x00000418 0x00004000    08%   0x00000010 EINTRPT 0x000000002022d1b0
erx      N/A   4   12  suspend 0x00000428 0x00004000    12%   0x00000010 EINTRPT 0x0000000020231b10
tsystem  N/A   4   30  suspend 0x000003f8 0x00002000    14%   0x00000020 EINTRPT 0x00000000202406e0
tidle3   N/A   3   31  ready   0x00000330 0x00001000    19%   0x00000020 OK      0x000000002023c5d8
tidle2   N/A   2   31  ready   0x00000330 0x00001000    19%   0x00000020 OK      0x000000002023c4d0
tidle1   N/A   1   31  ready   0x00000330 0x00001000    19%   0x00000020 OK      0x000000002023c3c8
tidle0   N/A   0   31  ready   0x00000360 0x00001000    41%   0x00000006 OK      0x000000002023c2c0
timer    N/A   4    4  suspend 0x00000410 0x00001000    29%   0x0000000a EINTRPT 0x0000000020242c08
msh />