RT-Thread/rt-thread #8983: qemu-vexpress-a9 中使用c++出错

messigogogo, 2024-05-24:
我在使用qemu-vexpress-a9 编译C++时出现以下错误，使用的编译链是arm-linux-musleabi_for_i686-w64-mingw32

[image]
[image]
[image]

messigogogo, 2024-05-24:
[image]

[image]

[image]

[image]

aarch64在开启C++后需要修改link.lds.S 中的__ctors_start   ->  __ctors_start__    ;   __ctors_end ->__ctors_end__
该编译链在pthread上定义有冲突，暂时删去了RT_USING_PTHREADS配置保证可以编译过。现象如上图所示

BernardXiong, 2024-05-24:
是不是也和工具链版本相关？另外，arm-linux-musleabi_for_x86_64-pc-linux-gnu 更多是和smart相关。可以多试试aarch64-none-elf工具链

messigogogo, 2024-05-27:
> 是不是也和工具链版本相关？另外，arm-linux-musleabi_for_x86_64-pc-linux-gnu 更多是和smart相关。可以多试试aarch64-none-elf工具链

您好，我刚才在64位试了一下aarch64-none-elf，没有发生以上问题，可以正常启动。
1.这个工具链是不支持C++11吗？开启C++11编译不过？如果要使用C++11,需要使用什么工具链？
2.rt-smat是否支持C++，使用aarch64-none-elf编译不过rt-smart。

messigogogo, 2024-05-27:
@BernardXiong

closed 2024-06-25 by messigogogo