RT-Thread/rt-thread #8998: rt_object_get_information函数优化问题

wdfk-prog, 2024-05-27:
- 这个函数应该调用的很频繁,应该可以优化,不需要for循环查找,直接从数组下标索引返回既可
- 源代码

```c
enum rt_object_class_type
{
    RT_Object_Class_Null          = 0x00,      /**< The object is not used. */
    RT_Object_Class_Thread        = 0x01,      /**< The object is a thread. */
    RT_Object_Class_Semaphore     = 0x02,      /**< The object is a semaphore. */
    RT_Object_Class_Mutex         = 0x03,      /**< The object is a mutex. */
    RT_Object_Class_Event         = 0x04,      /**< The object is a event. */
    RT_Object_Class_MailBox       = 0x05,      /**< The object is a mail box. */
    RT_Object_Class_MessageQueue  = 0x06,      /**< The object is a message queue. */
    RT_Object_Class_MemHeap       = 0x07,      /**< The object is a memory heap. */
    RT_Object_Class_MemPool       = 0x08,      /**< The object is a memory pool. */
    RT_Object_Class_Device        = 0x09,      /**< The object is a device. */
    RT_Object_Class_Timer         = 0x0a,      /**< The object is a timer. */
    RT_Object_Class_Module        = 0x0b,      /**< The object is a module. */
    RT_Object_Class_Memory        = 0x0c,      /**< The object is a memory. */
    RT_Object_Class_Channel       = 0x0d,      /**< The object is a channel */
    RT_Object_Class_ProcessGroup  = 0x0e,      /**< The object is a process group */
    RT_Object_Class_Session       = 0x0f,      /**< The object is a session */
    RT_Object_Class_Custom        = 0x10,      /**< The object is a custom object */
    RT_Object_Class_Unknown       = 0x11,      /**< The object is unknown. */
    RT_Object_Class_Static        = 0x80       /**< The object is a static object. */
};

struct rt_object_information *
rt_object_get_information(enum rt_object_class_type type)
{
    int index;

    type = (enum rt_object_class_type)(type & ~RT_Object_Class_Static);

    for (index = 0; index < RT_Object_Info_Unknown; index ++)
        if (_object_container[index].type == type) return &_object_container[index];

    return RT_NULL;
}
```

- 优化代码,直接从数组下标索引返回既可;枚举使用宏做变长

```c
enum rt_object_class_type
{
    RT_Object_Class_Null          = 0x00,      /**< The object is not used. */
    RT_Object_Class_Thread,      /**< The object is a thread. */
#ifdef RT_USING_SEMAPHORE
    RT_Object_Class_Semaphore,      /**< The object is a semaphore. */
#endif
    RT_Object_Class_Mutex       ,      /**< The object is a mutex. */
    RT_Object_Class_Event       ,      /**< The object is a event. */
    RT_Object_Class_MailBox     ,      /**< The object is a mail box. */
    RT_Object_Class_MessageQueue,      /**< The object is a message queue. */
    RT_Object_Class_MemHeap     ,      /**< The object is a memory heap. */
    RT_Object_Class_MemPool     ,      /**< The object is a memory pool. */
    RT_Object_Class_Device      ,      /**< The object is a device. */
    RT_Object_Class_Timer       ,      /**< The object is a timer. */
    RT_Object_Class_Module      ,      /**< The object is a module. */
    RT_Object_Class_Memory      ,      /**< The object is a memory. */
    RT_Object_Class_Channel     ,      /**< The object is a channel */
    RT_Object_Class_ProcessGroup,      /**< The object is a process group */
    RT_Object_Class_Session     ,      /**< The object is a session */
    RT_Object_Class_Custom      ,      /**< The object is a custom object */
    RT_Object_Class_Unknown     ,      /**< The object is unknown. */
    RT_Object_Class_Static               /**< The object is a static object. */
};

struct rt_object_information *
rt_object_get_information(enum rt_object_class_type type)
{
    int index;

    type = (enum rt_object_class_type)(type & ~RT_Object_Class_Static);

    if(type == RT_Object_Class_Null)
    {
        return RT_NULL;
    }
    else
    {
        return &_object_container[type - 1];
    }
}
```

- 不知这样是否可行?

BernardXiong, 2024-05-29:
不行呢，class_type是连续的，但是那个`_object_container`数组不是连续的：
```c
static struct rt_object_information _object_container[RT_Object_Info_Unknown] =
{
    /* initialize object container - thread */
    {RT_Object_Class_Thread, _OBJ_CONTAINER_LIST_INIT(RT_Object_Info_Thread), sizeof(struct rt_thread), RT_SPINLOCK_INIT},
#ifdef RT_USING_SEMAPHORE
    /* initialize object container - semaphore */
    {RT_Object_Class_Semaphore, _OBJ_CONTAINER_LIST_INIT(RT_Object_Info_Semaphore), sizeof(struct rt_semaphore), RT_SPINLOCK_INIT},
#endif
#ifdef RT_USING_MUTEX
    /* initialize object container - mutex */
    {RT_Object_Class_Mutex, _OBJ_CONTAINER_LIST_INIT(RT_Object_Info_Mutex), sizeof(struct rt_mutex), RT_SPINLOCK_INIT},
#endif
#ifdef RT_USING_EVENT
    /* initialize object container - event */
    {RT_Object_Class_Event, _OBJ_CONTAINER_LIST_INIT(RT_Object_Info_Event), sizeof(struct rt_event), RT_SPINLOCK_INIT},
#endif
#ifdef RT_USING_MAILBOX
    /* initialize object container - mailbox */
    {RT_Object_Class_MailBox, _OBJ_CONTAINER_LIST_INIT(RT_Object_Info_MailBox), sizeof(struct rt_mailbox), RT_SPINLOCK_INIT},
#endif
#ifdef RT_USING_MESSAGEQUEUE
    /* initialize object container - message queue */
    {RT_Object_Class_MessageQueue, _OBJ_CONTAINER_LIST_INIT(RT_Object_Info_MessageQueue), sizeof(struct rt_messagequeue), RT_SPINLOCK_INIT},
#endif
#ifdef RT_USING_MEMHEAP
    /* initialize object container - memory heap */
    {RT_Object_Class_MemHeap, _OBJ_CONTAINER_LIST_INIT(RT_Object_Info_MemHeap), sizeof(struct rt_memheap), RT_SPINLOCK_INIT},
#endif
...
```

wdfk-prog, 2024-05-29:
> ```c
> enum rt_object_class_type
> {
>     RT_Object_Class_Null          = 0x00,      /**< The object is not used. */
>     RT_Object_Class_Thread,      /**< The object is a thread. */
> #ifdef [...]

- `class_type`也用宏做区分，就可以和`_object_container`数组对应上了.
- 不知道`class_type`可以这样处理吗?我看了看代码,好像是没问题的

```c
enum rt_object_class_type
{
    RT_Object_Class_Null          = 0x00,      /**< The object is not used. */
    RT_Object_Class_Thread,      /**< The object is a thread. */
#ifdef RT_USING_SEMAPHORE
    RT_Object_Class_Semaphore,      /**< The object is a semaphore. */
#endif
#ifdef RT_USING_MUTEX
    RT_Object_Class_Mutex       ,      /**< The object is a mutex. */
#endif
#ifdef RT_USING_EVENT
    RT_Object_Class_Event       ,      /**< The object is a event. */
#endif
#ifdef RT_USING_MAILBOX
    RT_Object_Class_MailBox     ,      /**< The object is a mail box. */
#endif
#ifdef RT_USING_MESSAGEQUEUE
    RT_Object_Class_MessageQueue,      /**< The object is a message queue. */
#endif
#ifdef RT_USING_MEMHEAP
    RT_Object_Class_MemHeap     ,      /**< The object is a memory heap. */
#endif
  //其他一样的处理方式
    RT_Object_Class_MemPool     ,      /**< The object is a memory pool. */
    RT_Object_Class_Device      ,      /**< The object is a device. */
    RT_Object_Class_Timer       ,      /**< The object is a timer. */
    RT_Object_Class_Module      ,      /**< The object is a module. */
    RT_Object_Class_Memory      ,      /**< The object is a memory. */
    RT_Object_Class_Channel     ,      /**< The object is a channel */
    RT_Object_Class_ProcessGroup,      /**< The object is a process group */
    RT_Object_Class_Session     ,      /**< The object is a session */
    RT_Object_Class_Custom      ,      /**< The object is a custom object */
    RT_Object_Class_Unknown     ,      /**< The object is unknown. */
    RT_Object_Class_Static               /**< The object is a static object. */
};
```

nongxiaoming, 2024-05-29:
> * `class_type`也用宏做区分，就可以和`_object_container`数组对应上了.
> * 不知道`class_type`可以这样处理吗?我看了看代码,好像是没问题的
>
> ```c
> enum rt_object_class_type
> {
>     RT_Object_Class_Null          = 0x00,      /**< The object is not [...]

这个看着可以，但得每次修改都保证两边是对等的

BernardXiong, 2024-05-29:
为什么不这样做，是因为当这些修改了后，如果有些是以库的方式编译时，这些库整体会乱掉（因为class type不固定）。所以后来想到的办法时，class type是始终恒定的，而container数组是可以变动，然后就可以达到，没问题了。

而这个rt_object_get_information，应该调用次数并不算多

wdfk-prog, 2024-05-30:
> 为什么不这样做，是因为当这些内核对象menuconfig修改裁剪后，如果有些是以库的方式编译时，这些库整体会乱掉（因为class type不固定）。所以后来想到的办法是，class type是始终恒定的，而container数组是可以变动，然后就可以达到，没问题了。
> 
> 而这个rt_object_get_information，应该调用次数并不算多

行,有道理

closed 2024-05-30 by wdfk-prog