RT-Thread/rt-thread #9026: [Feature] 希望能将 formating 工具加到仓库的 tools 中

unicornx, 2024-06-04:
### Describe problem solved by the proposed feature

https://github.com/mysterywolf/formatting 这个工具建议直接加到仓库的 tools 目录下并添加文档说明，而不是每次 PR 的时候才知道原来有这个代码格式化工具。

Later-Comer, 2024-06-06:
目前仓亏支持clang-format自动格式的，一般的ide都支持clang-format，比如vscode只要配置下即可在保存的时候自动格式化。

unicornx, 2024-06-06:
> 目前仓库支持[clang-format](https://github.com/RT-Thread/rt-thread/blob/master/.clang-format)自动格式的，一般的ide都支持clang-format，比如vscode只要配置下即可在保存的时候自动格式化。

请问用这个 clang-format 的效果是否和 https://github.com/mysterywolf/formatting  的一致？如果使用 clang-format 是需要人手动在 ide 里配置的，如何保证大家的行动一致并符合 rtt 的要求呢？

Later-Comer, 2024-06-06:
> > 目前仓库支持[clang-format](https://github.com/RT-Thread/rt-thread/blob/master/.clang-format)自动格式的，一般的ide都支持clang-format，比如vscode只要配置下即可在保存的时候自动格式化。
> 
> 请问用这个 clang-format 的效果是否和 https://github.com/mysterywolf/formatting 的一致？如果使用 clang-format 是需要人手动在 ide 里配置的，如何保证大家的行动一致并符合 rtt 的要求呢？

两者功能有重叠，但是有些差异，其中https://github.com/mysterywolf/formatting 支持简单的代码格式化以及修改版权信息等功能，clang-format是强大的代码格式化工具，比如formatting强大很多，平时pr代码推荐使用clang-format。修改版权信息和日期是发布版本的事情。

unicornx, 2024-06-06:
也许是目前 github 上 pr 页面的描述误导了我。
[image]

Anyway，我个人觉得一个仓库自带的，立即可用的类似 linux 的 checkpatch 的工具也许会更方便大家检查。(formatting 工具还需要另外下载这个是我一开始觉得不方便的地方)

如果觉得我这个建议没有必要，请 feel free close this issue.

Regards.

BernardXiong, 2024-06-07:
感觉是有必要，可以考虑的。 @mysterywolf 觉得呢？

BernardXiong, 2024-06-07:
clang-format 是代码格式化，formating严格来说应该说是代码检查，因为提交代码的时候更多只能说是做代码检查的工作，而不是说帮助进行代码格式化。formating偏向机器人，所以机器人是不修改代码的。

dongly, 2024-06-27:
建议在仓库中,加入一个`.clang-format` 文件,提交 pr 前用 clang-format 代码化一来代码,这样代码风格就一致了

dongly, 2024-06-28:
没细看,原来已经有 .clang-format 文件了

closed 2025-09-02 by unicornx