RT-Thread/rt-thread #9043: [Bug] pm.c文件中函数_pm_change_sleep_mode()调用函数rt_timer_check()存在问题

youzuwei, 2024-06-06:
### RT-Thread Version

v5.1.0

### Hardware Type/Architectures

stm32l431-BearPi

### Develop Toolchain

Other

### Describe the bug

原因是：函数rt_timer_check()中有断言中断嵌套变量“RT_ASSERT(rt_interrupt_get_nest() > 0)”，然而函数_pm_change_sleep_mode()并不在中断中调用

wdfk-prog, 2024-07-08:
- 你有解决了嘛?
- 我看着也认为这是bug

将`rt_timer_check`移动到`rt_tick_set`后面既可;,这一块代码在中断屏蔽中执行,可以通过断言

```c
        /* enter lower power state */
        pm_sleep(pm, pm->sleep_mode);

        /* wake up from lower power state*/
        if (pm->timer_mask & (0x01 << pm->sleep_mode))
        {
            delta_tick = pm_lptimer_get_timeout(pm);
            pm_lptimer_stop(pm);
            if (delta_tick)
            {
                rt_tick_set(rt_tick_get() + delta_tick);
                rt_timer_check();
            }
        }

        /* resume all device */
        _pm_device_resume(pm->sleep_mode);

        if (_pm_notify.notify)
            _pm_notify.notify(RT_PM_EXIT_SLEEP, pm->sleep_mode, _pm_notify.data);

        rt_pm_exit_critical(level, pm->sleep_mode);
```

mysterywolf, 2024-07-20:
fixed: https://github.com/RT-Thread/rt-thread/pull/9172

closed 2024-07-20 by mysterywolf