RT-Thread/rt-thread #9049: [Bug] musl gcc can not work

unicornx, 2024-06-09:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek

### Develop Toolchain

Other

### Describe the bug

编译 RT-smart 的时候要求使用 musl 的 gcc，所以我从 <https://toolchains.bootlin.com/> 下载了一份 riscv64-lp64d--musl--stable-2024.02-1，但是编译的时候会报错.  build 的 log 见附件：
[build.log](https://github.com/user-attachments/files/15750401/build.log)

后来 google 发现 https://github.com/RT-Thread/rt-thread/issues/7594 上推荐了 <https://download.rt-thread.org/rt-smart/riscv64/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_180881.tar.bz2>,  这个就可以。

所以我的问题是 RT-smart 对 musl gcc 有什么特殊要求吗？我们是不是只能用 RTT 官方的 toolchain？

a1012112796, 2024-06-09:
是的，只能用rtt 自己的,估计因为rtsmart 的系统调用等接口未必兼容其他系统。建议官方直接把文件名改为 `riscv64-rtsmart-musleabi_for_x86_64-...`， 这样应该会更好理解一些。

unicornx, 2024-06-09:
好的，非常感谢。

请确认一下，我上面列的 <https://download.rt-thread.org/rt-smart/riscv64/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_180881.tar.bz2> 是否就是 RTT 官方的 musl gcc？

BernardXiong, 2024-06-10:
是的，是需要对应的版本，而不是完全musl gcc的工具链。

BernardXiong, 2024-06-10:
180881这个版本可能还有更新的版本，目前不完全确认，也包括riscv64上还在修复一些问题

unicornx, 2024-06-10:
> 180881这个版本可能还有更新的版本，目前不完全确认，也包括riscv64上还在修复一些问题

@BernardXiong 请问 RT-Thread 是否有发布和列出所有的 riscv 的 musl 的 gcc 的官方链接？<https://download.rt-thread.org/rt-smart/riscv64/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_180881.tar.bz2> 这个只是对应某一个的，我希望有个网页列出了所有的版本以及最新的版本，这样也方便我们自己去下载。

BernardXiong, 2024-06-13:
因为这些工具链版本都是自动产生的，编号会自动递增，这个链接会指向最新版本：

https://download-redirect.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2

unicornx, 2024-06-13:
> 因为这些工具链版本都是自动产生的，编号会自动递增，这个链接会指向最新版本：
> 
> https://download-redirect.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2

thanks， 所以我们平时只要用这个最新的版本就好了，是吗？并没有什么 stable 版本的说法？

BernardXiong, 2024-06-13:
> > 因为这些工具链版本都是自动产生的，编号会自动递增，这个链接会指向最新版本：
> > https://download-redirect.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2
> 
> thanks， 所以我们平时只要用这个最新的版本就好了，是吗？并没有什么 stable 版本的说法？

是的，大部分场合用这份就可以了。目前还没发带版本戳的stable版本

unicornx, 2024-06-24:
该 issue 已经基本解决。close now

closed 2024-06-24 by unicornx

fengluochaun, 2024-07-11:
有windows能用的版本连接么？