RT-Thread/rt-thread #9063: [Bug] Not compiled for Allwinner D1H.

ylyamin, 2024-06-13:
### RT-Thread Version

master, v5.1.0, v5.0.2

### Hardware Type/Architectures

Sipeed Lichee RV board

### Develop Toolchain

Other

### Describe the bug

I like this project but can't build for Allwinner D1H on Sipeed Lichee RV board

I use compiler https://github.com/RT-Thread/toolchains-ci/releases/download/v1.7/riscv64-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2

I was trying:

On [Master](https://github.com/RT-Thread/rt-thread/commit/2866da37a02dae72200e02ad87480718002760a5) or [v5.1.0](https://github.com/RT-Thread/rt-thread/releases/tag/v5.1.0):
- bsp/allwinner/d1/ - not compiled looks outdated
- bsp/allwinner/d1s/ - not compiled  

On [v5.0.2](https://github.com/RT-Thread/rt-thread/releases/tag/v5.0.2):  
- bsp/allwinner/d1/ - not compiled looks outdated
- bsp/allwinner/d1s/ - compiled, but no output to UART. 

After I forked v5.0.2 and added:
- bsp\allwinner\d1s\tools\boot0_sdcard_sun20iw1p1_d1s.bin change to varian from https://github.com/smaeul/sun20i_d1_spl.git
- bsp\allwinner\d1s\sbi.bin change to varian from https://github.com/riscv-software-src/opensbi
- Change UART pins and boudrate in config 

After this changes board started and shown out to UART.
But loading frosen in rt_thread_mdelay function in files
- bsp/allwinner/libraries/drivers/sdmmc/drv_sdmmc.c
- bsp/allwinner/libraries/sunxi-hal/include/hal/sdmmc/osal/RT-Thread/_os_time.h

Chnage it to rt_hw_us_delay and is finnaly looks like works:
```
 \ | /
- RT -     Thread Smart Operating System
 / | \     5.0.2 build Jun  9 2024 17:11:05
 2006 - 2022 Copyright by RT-Thread team
hal_sdc_create 0
card_detect insert
Initial card success. capacity :15200MB
sdmmc bytes_per_secotr:200, sector count:1db0000
not found partition of mbr, construct sd0 at offset 8M, size:0xffffffffb5800000
[D/FAL] (fal_flash_init:47) Flash device |                  sdcard0 | addr: 0x00000000 | len: 0xb6000000 | blk_size: 0x00000200 |initialized finish.
[I/FAL] ==================== FAL partition table ====================
[I/FAL] | name       | flash_dev |   offset   |    length  |
[I/FAL] -------------------------------------------------------------
[I/FAL] | download   | sdcard0   | 0x00800000 | 0x00800000 |
[I/FAL] | easyflash  | sdcard0   | 0x01000000 | 0x00100000 |
[I/FAL] | filesystem | sdcard0   | 0x01100000 | 0x00c00000 |
[I/FAL] =============================================================
[I/FAL] RT-Thread Flash Abstraction Layer initialize success.
Hello RISC-V
msh />Mount "sd0p0" on "/" fail
msh />
msh />
```
This my changes https://github.com/RT-Thread/rt-thread/commit/ec2c0f10894bca703354e7e1d4c23fbaed5a034b

Please could you make it compiled properly in master and figure out what happened with SDMMC driver?

polarvid, 2024-07-04:
This BSP has been unmaintained since v5.0.0 I believe 😢

Recent works focused more on milk-v(https://github.com/RT-Thread/rt-thread/tree/master/bsp/cvitek) according to the git log.

ylyamin, 2024-07-05:
Hi @polarvid, thank you for your response.
I can maintain D1 bsp, can create pull requests. 
But since I’m not an expert in RTT, someone needs to review it. My current solution to just replace thread_delay to hw_delay doesn't seem quite right.

polarvid, 2024-07-05:
I have a D1s board to help with the kernel compiling issue on master branch and do some testing on it.

While I'm not an expert in sdhci driver, Im afraid there's not much I can do... 🤔Or you could submit your modification as a PR and wait for more experienced contributors to review it.

ylyamin, 2024-07-05:
Hi, @polarvid
It would be great if you could fix the compiling of bsp/allwinner/d1s/ on Master,  to keep D1 BSP updated in Master. Thank you for interest.
(I think bsp/allwinner/d1/ folder - looks very old and I believe it can be deleted, bsp/allwinner/d1s/ can be one folder for D1s and D1H)

About sdhci yes looks like needs experts, but with my dirty fix it works, I can do PR.
Want to say actually right now I achieved working LCD RGB and MIPI DCI Display with D1H and RTT, but completely stuck on how to up USB Host function. Host driver initiated but the thread is suspended and nothing happens.

ylyamin, 2024-07-05:
Thank you for the nice words @polarvid .

> you could try the backtrace() to find out where thread suspended or dump more status on it for debugging

Yes, I tried backtrace() still can't understand what happened. Thinking maybe instead RTT driver try TinyUSB. Believe I need to create a separate git issue for this topic.

> please feel free to share any details or updates whenever you can

I believe now all that was done required a big blog article, maybe I do it someday. Don't know how many people are interested in D1 drivers. Right now I have rough notes: https://github.com/ylyamin/RT-Thread-on-Allwinner-D1H/blob/master/documentation/Code_changes.md

polarvid, 2024-07-06:
Hi 😄@ylyamin

I had tested the bringup and filesystem in sd on my D1s board, with the latest kernel on my dev branch (on #9142). And to summary, besides the compiling failure, the sd driver works just fined with me.

[image]

My log is here:

```text

 \ | /
- RT -     Thread Smart Operating System
 / | \     5.2.0 build Jul  6 2024 13:41:01
 2006 - 2024 Copyright by RT-Thread team
hal_sdc_create 0
card_detect insert
Initial card success. capacity :121MB
sdmmc bytes_per_secotr:200, sector count:3c800
found part[0], begin: 8388608, size: 90.0MB
found partition:sd0 of mbr at offset 0x0000000000004000, size:0x000000000002d000
found part[1], begin: 102760448, size: 23.0MB
found partition:sd1 of mbr at offset 0x0000000000031000, size:0x000000000000b800
hal_sdc_create 1
card_detect insert
Initial card success. capacity :29818MB
sdmmc bytes_per_secotr:200, sector count:3a3d000
found part[0], begin: 4194304, size: 256.0MB
found partition:sd0 of mbr at offset 0x0000000000002000, size:0x0000000000080000
found part[1], begin: 272629760, size: 1.716GB
found partition:sd1 of mbr at offset 0x0000000000082000, size:0x0000000000366000
found part[2], begin: 2097152000, size: 27.169GB
part_count1:2
found partition:sd2 of mbr at offset 0x00000000003e8000, size:0x0000000003654800
[D/FAL] (fal_flash_init:47) Flash device |                  sdcard0 | addr: 0x00000000 | len: 0x07900000 | blk_size: 0x00000200 |initialized finish.
[I/FAL] ==================== FAL partition table ====================
[I/FAL] | name       | flash_dev |   offset   |    length  |
[I/FAL] -------------------------------------------------------------
[I/FAL] | download   | sdcard0   | 0x00800000 | 0x00800000 |
[I/FAL] | easyflash  | sdcard0   | 0x01000000 | 0x00100000 |
[I/FAL] | filesystem | sdcard0   | 0x01100000 | 0x00c00000 |
[I/FAL] =============================================================
[I/FAL] RT-Thread Flash Abstraction Layer initialize success.
Hello RISC-V
msh />[W/time] Cannot find a RTC device!
Mount "sd0p0" on "/" success
[E/DFS.fs] mount point (/) already mounted!
Mount "sd0p1" on "/data" fail
hal_sdc_create 1
[os E] OS_MutexCreate():42, handle 0x4055c040
card_detect insert
card_detect not eixst
sdmmc bytes_per_secotr:200, sector count:3a3d000
found part[0], begin: 4194304, size: 256.0MB
found partition:sd0 of mbr at offset 0x0000000000002000, size:0x0000000000080000
found part[1], begin: 272629760, size: 1.716GB
found partition:sd1 of mbr at offset 0x0000000000082000, size:0x0000000000366000
found part[2], begin: 2097152000, size: 27.169GB
part_count1:2
found partition:sd2 of mbr at offset 0x00000000003e8000, size:0x0000000003654800
Mount "sd1p0" on "/mnt" success

msh />
msh />ls
Directory /:
system              <DIR>
download            <DIR>
mnt                 <DIR>
data                <DIR>
entries.log         110303
smart-ksymtbl.txt   154545
dev                 <DIR>
msh />
msh />cat smart-ksymtbl.txt
0x00000000404033e6 hal_hw_pin_init
0x0000000040403410 hal_pin_irq_enable
0x0000000040403496 hal_pin_detach_irq
0x000000004040350e hal_pin_attach_irq
0x00000000404035aa hal_pin_read
0x00000000404035d6 hal_pin_write
0x00000000404035f4 hal_pin_mode
0x0000000040403688 _uart_putc
0x00000000404036cc _uart_getc
```

ylyamin, 2024-07-06:
Hi, @polarvid 

Amazing work, thank you very much for your efforts ! I hope PR will be approved. I am happy that D1 bsp became alive in RTT.
Could you tell me how you handled the compilation failure, it is known how to fix it?

polarvid, 2024-07-06:
@ylyamin

And I read your notes on https://github.com/ylyamin/RT-Thread-on-Allwinner-D1H/blob/master/documentation/Code_changes.md, which is incredibly insightful to me. They can be edited and posted to the forum to benefit others in the community if you would love to 😃

ylyamin, 2024-07-06:
Hi @polarvid,

Great, so your solve D1S compilation is very good, thanks !
About D1H support, maybe need to suggest RTT masters to use the common D1s-D1H folder. I can create PR to delete old folders and add bootloaders, uart etc. Or you can do PR, I'm a newbie in this. After this, this original issue will be completely solved.

By the way, in your PR proposal to delete rt_hw_us_delay(), I used this function to solve LCD driver frize in my platform, need to think about it.

About Code_changes.md. Yes I want to add more human explanations and pictures and can be published in forum, hope to work with it after fighting with USB.

polarvid, 2024-07-06:
>  delete rt_hw_us_delay()

It seems that API has been implemented on libcpu/risc-v/t-head/c906, so that won't be a problem for your modifications. But with my test on master, the sd seems fine. I am not sure if that will also work with you on d1h board. You could try it on later on master if possible.

 > the common D1s-D1H folder

I think it could be better if you open a PR on it. Since the proporsal came from you. And I'm willing to help if there is any problem.

ylyamin, 2024-07-06:
Understand and agree, thank you for your support!

closed 2024-07-09 by ylyamin

ylyamin, 2024-10-18:
Hi, @polarvid,

If you remember our conversation about RTT and D1H. Sorry I comment in closed issue.

> "And I read your notes which is incredibly insightful to me. They can be edited and posted to the forum to benefit others in the community if you would love to 😃"

I create series of articles covering the knowledge and experience I’ve gained: https://github.com/ylyamin/RT-Thread-on-Allwinner-D1H/blob/master/documentation/D1_0_index.md

And share this in forum: https://club.rt-thread.org/ask/article/b66f7db40179a81b.html, https://club.rt-thread.io/ask/article/18.html

However with my D1H project - I still stuck with the USB driver. It can detect device connections, but communication has not started.

polarvid, 2024-10-21:
Hi, @ylyamin!

I have read your articles for the D1H board. They are pretty comprehensive!

And if you'd like to, you could provide your email address, so we can easily exchange information more (BTW, you could reach out to me at smokewood@qq.com).

ylyamin, 2024-10-21:
Hi, @polarvid 
Thank you for your reply. Yes, my address is yrans1@gmail.com