RT-Thread/rt-thread #9075: [Bug] bsp/cvitek 大核无法支持 float 指令

unicornx, 2024-06-20:
### RT-Thread Version

master

### Hardware Type/Architectures

bsp/cvitek

### Develop Toolchain

GCC

### Describe the bug

bsp/cvitek/cv18xx_risc-v/applications/main.c 的 main 中加入如下代码：
```c
flaot f = 100;
```
编译正常，但是运行会 crash，报 Unhandled Exception 2:Illegal Instruction

同样的代码加在 bsp/cvitek/c906_little/applications/main.c 中就正常

简单分析后发现大小核的编译选项有区别：
大核：-mabi=lp64

小核：-mabi=lp64d

统一成 -mabi=lp64d， 大核仍然 crash

后继续将大核的 gcc 换成 musl 的，即 https://download-redirect.rt-thread.org/download/rt-smart/toolchains/riscv64gc-linux-musleabi_for_x86_64-pc-linux-gnu_latest.tar.bz2

发现链接时报错：can't link soft-float modules with double-float modules

unicornx, 2024-06-24:
走读代码时发现 `libcpu/risc-v/t-head/c906/startup_gcc.S` 存在如下逻辑：

```asm
    /* set to disable FPU */
    li t0, SSTATUS_FS
    csrc sstatus, t0
    li t0, SSTATUS_SUM
    csrs sstatus, t0
```

怀疑这会影响 float 指令的执行，但注释掉这四行语句后，运行依然报非法指令错误。

所以想问一下，当时为啥要 diable FPU？

unicornx, 2024-06-24:
@BernardXiong 搜索 `libcpu/risc-v/t-head/c906/` 代码，还发现有 `ifdef ENABLE_FPU` 的逻辑，但是 Kconfig 中并没有 ENABLE_FPU 的配置开关。

请问当时移植 `libcpu/risc-v/t-head/c906/` 的代码历史是什么？感觉需要再整理一下？

unicornx, 2024-06-24:
强制在 rtconfig.h 中定义 ENABLE_FPU。保留  `libcpu/risc-v/t-head/c906/startup_gcc.S` 中 diable FPU 的操作。编译ok，运行也正常，但是发现 kprintfk 无法打印 %f

```c
int main(void)
{
    int i= 100;
    float f = i;

    float x = f / 2;
    i = x;

    rt_kprintf("x = %f\n", x);
    rt_kprintf("Hello RTT test!, x = %d\n", i);
    return 0;
}
```

运行显示：
```shell
x = %f
Hello RTT test!, x = 50
```

unicornx, 2024-06-24:
目前这个 issue 实际上可以分成两个问题
- 问题 1：基于 newlib gcc 的 FPU 没有 enable 导致 float 指令异常
- 问题 2：将 `-mabi=lp64` 改成 `-mabi=lp64d` 后换成 musl gcc 链接报错。

从长远看，如果大核要上 smart，则要用 musl gcc，但是由于 问题 2 ，可能要从 toolchain 角度再看看，这个应该要用另外一个 issue 跟踪。

如果单纯为了解决问题 1，不换 newlib gcc，那么我认为可以采用修改 Kconfig，增加 `ENABLE_FPU` 选项方式。但 @flyingcys 建议采用 `ARCH_RISCV_FPU_D` 更好。

[image]

polarvid, 2024-07-09:
> 走读代码时发现 `libcpu/risc-v/t-head/c906/startup_gcc.S` 存在如下逻辑：
> 
> ```assembly
>     /* set to disable FPU */
>     li t0, SSTATUS_FS
>     csrc sstatus, t0
>     li t0, SSTATUS_SUM
>     csrs sstatus, t0
> ```
> 
> 怀疑这会影响 float 指令的执行，但注释掉这四行语句后，运行依然报非法指令错误。

FPU VPU 是按需开启的 😂 否则会拖垮大部分常规任务的整体性能

unicornx, 2024-07-09:
> > 目前这个 issue 实际上可以分成两个问题
> > 
> > * 问题 1：基于 newlib gcc 的 FPU 没有 enable 导致 float 指令异常
> > * 问题 2：将 `-mabi=lp64` 改成 `-mabi=lp64d` 后换成 musl gcc 链接报错。
> > 
> > 从长远看，如果大核要上 smart，以及为了解决 #8943，则要用 musl gcc，但是由于 问题 2 ，可能要从 toolchain 角度再看看，这个应该要用另外一个 issue 跟踪。
> 
> 针对问题 2 新提了一个 issue：#9094。本 issue 仅用于跟踪 问题 1。

问题 2 应该不是问题，保留 `-mabi=lp64` 即可。#9094 已经关闭。

unicornx, 2024-07-09:
参考 https://github.com/RT-Thread/rt-thread/blob/master/bsp/qemu-virt64-riscv/README_ch.md

> * 如果 `-march` 包含 `v`, 选择 `Using RISC-V Vector Extension`, 反之关闭
> * 如果 `-march` 包含 `f/d`, 选择 `Enable FPU`, 反之关闭

unicornx, 2024-08-12:
#9290 为 cvitek 大核使能了 float，如果这个 pr 能进主线，也就解决了本 issue。tracking......

mysterywolf, 2024-08-24:
https://github.com/RT-Thread/rt-thread/pull/9290 这个要合并吗？

unicornx, 2024-08-25:
> #9290 这个要合并吗？

先 review 吧 ~~~

mysterywolf, 2024-08-27:
solved: https://github.com/RT-Thread/rt-thread/pull/9290

closed 2024-08-27 by mysterywolf

unicornx, 2024-09-04:
verified float can work now after #9290  is merged.
Verified commit id: d2160d352c