RT-Thread/rt-thread #9141: [Bug]

Rbb666, 2024-07-06:
### RT-Thread Version

4.x 5.0.0 5.0.2

### Hardware Type/Architectures

none

### Develop Toolchain

MDK Keil

### Describe the bug

在2.0版本env-windows工具测试生成mdk工程出现的问题：
* 内核版本：经测试除了当前 master 分支，4.x、5.0.x、5.1.0 版本内核都会这个问题
[image]
* 分析：主要原因是env2.0升级了py3版本，部分语法和py3不兼容：decode 函数，py3由于字符串已经是utf8了不需要再次decode。如果修改，则所有版本rtt仓库都需要调整 tools/keil iar....py 文件
[image]

mysterywolf, 2024-07-06:
这个可以考虑5.2.0使用env2.0，其他版本继续使用env 1.5.x

Later-Comer, 2024-07-07:
> 这个可以考虑5.2.0使用env2.0，其他版本继续使用env 1.5.x

估计是一种比较好的解决思路，这样env2.0没有历史包袱